PhilipsHue/flutter_reactive_ble #339: WriteCharacteristic doesn't work stable on IOS

gtggggggggggggggggggggggggggggggggggggg, 2021-08-11:
I work on an App which works properly on Android, but does not on IOS. There are 3 Exceptions i receive mainly.

This Exception reveals that the **deviceID** which I have saved from the scan-process is unknown despite the first 2 `WriteCharacteristicWithResponse` commands, where I use the same deviceID,  don't throw any Exceptions.

`Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "A peripheral <deviceID> is unknown (make sure it has been discovered)")`

Further on I often get this Exception. It's not like there's a problem with the characteristic, because on my first 2 commands it works and also on Android it works.

`Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "The characteristic <CharacteristicID> of the service <ServiceID> of the peripheral <deviceID> is not writable")`

And the most annoying Exception is the PlatformException. The only thing it tells is that something doesn't work. After a lot of research I still don't know how to handle this Exception, not to mention how to prevent it.

`[VERBOSE-2:ui_dart_state.cc(186)] Unhandled Exception: PlatformException(flutter_reactive_ble.Central.(unknown context at $100e1e000).Failure:1, The operation couldn’t be completed. (flutter_reactive_ble.Central.(unknown context at $100e1e000).Failure error 1.), {}, null)
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:581:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:158:18)
<asynchronous suspension>
#2      new Stream.fromFuture.<anonymous closure> (dart:async/stream.dart)
<asynchronous suspension>
`

**Smartphone / tablet**
 - Device: iPhone XS, iPhone 6
 - OS: IOS 12.5.4, IOS 14.
 - Package version: 3.1.1+1

**Peripheral device**
- Vendor, model: device from my company
- Does it run a custom firmware: yes

**Additional context**
On IOS it simply does not work and there's nearly no documentation to find. There are a few issues that describe similar problems, but no real solution is provided. And the worst part is that the Exceptions don't describe the problems at all.

The issue [https://github.com/PhilipsHue/flutter_reactive_ble/issues/291](url) describes one of my problems and there is not really a solution for the problem.

remonh87, 2021-08-13:
@gtggggggggggggggggggggggggggggggggggggg  can you check if you experience the same with the example app?

gtggggggggggggggggggggggggggggggggggggg, 2021-08-13:
@yuriboeira11tx no, when I get an Exception nothing works and the device doesn't get any data

gtggggggggggggggggggggggggggggggggggggg, 2021-08-13:
@remonh87  unfortunately I'm on vacation now, I can test it in 10 days

remonh87, 2021-08-14:
@yuriboeira11tx  can you test if you experience the same using our example app?

yuriboeira11tx, 2021-08-16:
> @yuriboeira11tx você pode testar se você experimenta o mesmo usando nosso exemplo de aplicativo?

Honestly, I can't even build the sample app for iOS. There is a problem with the 8.0 set version, the minimum is 9.0.

```
IPHONEOS_DEPLOYMENT_TARGET is set to 8.0, but the range of supported deployment target versions is 9.0 to 14.0.99
```

Taym95, 2021-08-23:
> > @yuriboeira11tx você pode testar se você experimenta o mesmo usando nosso exemplo de aplicativo?
> 
> Honestly, I can't even build the sample app for iOS. There is a problem with the 8.0 set version, the supported is 9.0.
> 
> ```
> IPHONEOS_DEPLOYMENT_TARGET is set to 8.0, but the range of supported deployment target versions is 9.0 to 14.0.99
> ```

This happens because support for iOS 8 has been dropped, but the minimum deployment target for the pod is iOS 8, add this to your ios Podfile;

`post_install do |installer|
  installer.pods_project.targets.each do |target|
    target.build_configurations.each do |config|
      config.build_settings.delete 'IPHONEOS_DEPLOYMENT_TARGET'
    end
  end
end
`
Give it a try and let us know, please!

gtggggggggggggggggggggggggggggggggggggg, 2021-08-23:
@remonh87 

With the example app it works solid, only thing is that I get

`flutter: Error unsubscribing from notifications: PlatformException(flutter_reactive_ble.Central.(unknown context at $1016ec0e0).Failure:1, The operation couldn’t be completed. (flutter_reactive_ble.Central.(unknown context at $1016ec0e0).Failure error 1.), {}, null)`

everytime the device disconnects, but I guess that's ok for me.

But I still don't understand why I get the errors 'device_id is unknown' and 'characteristic not writeable' despite using the same devices as with the demo app and also my code is basically the same.

Taym95, 2021-08-23:
> @remonh87
>
> With the example app it works solid, only thing is that I get
>
> `flutter: Error unsubscribing from notifications: PlatformException(flutter_reactive_ble.Central.(unknown context at [...]

That's okay because every time device disconnects it unsubscribing from notifications

gtggggggggggggggggggggggggggggggggggggg, 2021-08-24:
@Taym95 Do you know where I can handle the error?

closed 2021-09-06 by gtggggggggggggggggggggggggggggggggggggg

farr64, 2021-09-10:
@Taym95 Commented:

> That's okay because every time device disconnects it unsubscribes from notifications, even though should not show an error!

This is a message from the Library. Please add logic to the Library along these lines to eliminate the confusing console message:

If a device is not connected, don't unsubscribe from notifications becase we **know** that attempting to do anything with the device **WILL** fail and this (predictable failure) **WILL** generate a spurious message.

In general: If a device is not connected, don't do anything with it.

Thanks