PhilipsHue/flutter_reactive_ble #342: Cannot connect to a previously paired BLE device on iOS (crashes after failures)

calebisstupid, 2021-08-17:
**Describe the bug**
I have a push to talk ble device, and I am able to successfully scan for it, connect to it, and subscribe to characteristics without issue. However, if I turn on discovery/pairing on the ble device again, any future attempts to connect to this device result in assertion failures and an app crash.

If I open the iOS bluetooth settings and choose "Forget this device", everything works fine again until I attempt to reconnect.

This is not happening on android.

I notice in the documentation for this device, that is uses some sort of streamlined pairing. I have not build anything with ble devices that have this before, so maybe there is some easy fix that I am not aware of. It claims you can just pair it to the phone once and it is "good to go", but I am at a loss trying to find out why I cannot re-connect to it after discovering it again.

Logs from the example app:
```
fopen failed for data file: errno = 2 (No such file or directory)
Errors found! Invalidating cache...
fopen failed for data file: errno = 2 (No such file or directory)
Errors found! Invalidating cache...
Warning! No event channel set up to report a connection update
2
flutter_reactive_ble/ConnectTaskController.swift:16: Assertion failed
* thread #1, queue = 'com.apple.main-thread', stop reason = Assertion failed
    frame #0: 0x00000001a0ffaeac libswiftCore.dylib`_swift_runtime_on_report
libswiftCore.dylib`_swift_runtime_on_report:
->  0x1a0ffaeac <+0>: ret
libswiftCore.dylib`_swift_reportToDebugger:
    0x1a0ffaeb0 <+0>: b      0x1a0ffaeac               ; _swift_runtime_on_report
libswiftCore.dylib`_swift_shouldReportFatalErrorsToDebugger:
    0x1a0ffaeb4 <+0>: adrp   x8, 337380
    0x1a0ffaeb8 <+4>: ldrb   w0, [x8, #0x299]
Target 0: (Runner) stopped.
Lost connection to device.
```

**To Reproduce**
Steps to reproduce the behavior (using example project):
1. Scan, connect, and test example app using a ble push to talk device.
2. Re-enable pairing/discovery on ble device
3. Scan for device, and find it
4. Attempt to connect to the device
5. See a handful of timeouts or assertion failures
6. App crashes

**Expected behavior**
Can discover and connect to a device that has been previously paired.

**Smartphone / tablet**
 - Device: iPhone 11
 - OS: 14.7.1
 - Package version: 3.1.1
 
 (also tested on an iPhone X)

**Peripheral device**
- [https://www.ainaptt.com/ptt-smart-button/](https://www.ainaptt.com/ptt-smart-button/)
- Does it run a custom firmware / software: no

remonh87, 2021-08-18:
@werediver  you have a bit more knowledge then I regarding the iOS implementation. Can you have a look at it?

austinn, 2021-09-23:
I am also experiencing this crash when connecting to a device that is not connected. iOS only

closed 2021-11-11 by remonh87