PhilipsHue/flutter_reactive_ble #344: package com.signify.hue.flutterreactiveble does not exist (Android)

rohitranjan1991, 2021-08-19:
**Describe the bug**
I am getting the following issue while building my code with this plugin added in pubspec.yaml for Android.

Error: 

`GeneratedPluginRegistrant.java:35: error: package com.signify.hue.flutterreactiveble does not exist
    flutterEngine.getPlugins().add(new com.signify.hue.flutterreactiveble.ReactiveBlePlugin());`

**To Reproduce**
Steps to reproduce the behavior:
1. Add 4.0.0 version of this library
2. Try to run the code on a device
3. The build fails with the above exception

**Expected behavior**
The build should work

**Smartphone / tablet**
 - Device: Android
 - OS: Android 10
 - Package version: 4.0.0

remonh87, 2021-08-19:
Did you follow the instructions provided:  https://github.com/PhilipsHue/flutter_reactive_ble#setup ?

rohitranjan1991, 2021-08-20:
@remonh87 , Yes I did. I have set all the required permissions. Also, I have set the proguard rule also. Please let me know what else could I be doing wrong. Thanks.

remonh87, 2021-08-28:
I tried the same and for me it works without issues. You probably also did a flutter clean and rebuild the app?

closed 2021-09-09 by remonh87