PhilipsHue/flutter_reactive_ble #345: BLE undeliverable exception - need more documentation on how to implement it

chris-otani, 2021-08-19:
I'm struggling to add the workaround for the BLE undeliverable exception to my flutter project. I have no Java/Kotlin experience and searched but can't figure out how to properly add it.

I've added the code to my MainActivity.kt but am getting errors. Could you provide more detail on where to add this code?

> error: expecting member decleration

[image]

When I try to run it I get:

> Unresolved reference: throwable
> FAILURE: Build failed with an exception.

`
import io.flutter.embedding.android.FlutterActivity

class MainActivity: FlutterActivity() {
    RxJavaPlugins.setErrorHandler { throwable ->
        if (throwable is UndeliverableException && throwable.cause is BleException) {
            return@setErrorHandler // ignore BleExceptions since we do not have subscriber
        }
        else {
            throw throwable
        }
    }
}`

remonh87, 2021-08-20:
Do not forget to add

`implementation "com.polidea.rxandroidble2:rxandroidble:1.11.1"` to your gradle dependencies and fix the imports

chris-otani, 2021-08-20:
@remonh87 , thanks for the quick response.

I added that to my gradle dependicies. I also added rxkotlin since it couldn't find that package.

```
dependencies {
    implementation "org.jetbrains.kotlin:kotlin-stdlib-jdk7:$kotlin_version"
    implementation "com.polidea.rxandroidble2:rxandroidble:1.11.1"
    implementation 'io.reactivex.rxjava2:rxkotlin:2.4.0'
}
```

When fixing the imports it looks like It's not properly importing the "RxJavaPlugins".  the import is greyed out like its not being used

MainActivity.kt
```
import io.flutter.embedding.android.FlutterActivity

import com.polidea.rxandroidble2.exceptions.BleException
import io.reactivex.exceptions.UndeliverableException
import io.reactivex.plugins.RxJavaPlugins

class MainActivity: FlutterActivity() {
    RxJavaPlugins.setErrorHandler { throwable ->
        if (throwable is UndeliverableException && throwable.cause is BleException) {
            return@setErrorHandler // ignore BleExceptions since we do not have subscriber
        }
        else {
            throw throwable
        }
    }
}
```

Here are the static errors:

[image]

keval-devani, 2021-08-25:
Just change your code like this in your MainActivity
```
import android.os.Bundle
import com.polidea.rxandroidble2.exceptions.BleException
import io.flutter.embedding.android.FlutterActivity
import io.reactivex.exceptions.UndeliverableException
import io.reactivex.plugins.RxJavaPlugins

class MainActivity : FlutterActivity() {

    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)

        RxJavaPlugins.setErrorHandler { throwable ->
            if (throwable is UndeliverableException && throwable.cause is BleException) {
                return@setErrorHandler // ignore BleExceptions since we do not have subscriber
            } else {
                throw throwable
            }
        }
    }
}

```

closed 2021-08-28 by remonh87