PhilipsHue/flutter_reactive_ble #346: ScanFilter

keval-devani, 2021-08-24:
**Is your feature request related to a problem?**
I developing app that scan and received only advertising data for that I just need to scan device no need to connect with device. But, When screen is off app is stop scan device and I search some answers on stackoverflow I need to pass ScanFilter to continue scan even screen is off. 

**Describe the solution you'd like**
Add manufacturerData option like withServices in scanForDevices method so I can pass filer.  

**Additional context**
for reference: https://newbedev.com/ble-scan-is-not-working-when-screen-is-off-on-android-8-1-0

keval-devani, 2021-08-28:
My issue is I want to pass manufacturer ID on start scan not service uuid. That way I can receive ble device even on screen off. Please check this https://stackoverflow.com/a/48079800

remonh87, 2021-08-28:
We do not support this as off now. Also I need to check if iOS provides a filter for scans based on manufacturer data.

 So as workaround you can filter on a serviceuuid for now (which avoids the issue in stack overflow) and then to filter out your devices you can use the stream where operator.