PhilipsHue/flutter_reactive_ble #351: isScanning?

avdisx, 2021-08-31:
how can i set the always non-stop scan in the package. According to my experiments, there is no problem on the iOS platform, but on Android it stops scanning after a short time. We could track it from the isScanning variable in the flutter_blue package, which unfortunately does not. how can I do it.
Have a nice day

remonh87, 2021-09-02:
in general Android kills off bluetooth scans which is by default 30 minutes if I recall this correctly. It can be that the specific BLE device implementation stops scanning sooner. However I would expect `scanForDevices` stream would complete. What you can do is write some business logic yourself (e.g. by using the `Ondone` callback) to reinitiate the scan

avdisx, 2021-09-02:
> in general Android kills off bluetooth scans which is by default 30 minutes if I recall this correctly. It can be that the specific BLE device implementation stops scanning sooner. However I would [...]

but its 4 minutes

closed 2021-09-18 by remonh87