PhilipsHue/flutter_reactive_ble #352: Any plans to support BLE5 functions?

lcrocker, 2021-08-31:
Specifically, Android API 26 BluetoothGatt.setPreferredPhy() and readPhy().

If anyone thinks they could implement these in a reasonable time frame, I could probably arrange for it to be a paying gig.

remonh87, 2021-09-02:
I need to check if iOS exposes these functions as well if so we can consider adding it to the library.

lcrocker, 2021-09-03:
It's part of the BLE standard, and Android apps are using it now. Frankly, if Apple hasn't caught up yet, that's their problem.

lcrocker, 2021-09-17:
FYI, I have offered a bounty on this question:

https://stackoverflow.com/questions/69184440/replacing-a-gradle-dependency-in-flutter-library

wldeveloperapps, 2022-08-25:
Hi, any advance in this matter, I need to link BLE 5.0 devices with the LongRange feature...

azhansy, 2024-04-17:
Polidea  SDK support readPhy() already.