PhilipsHue/flutter_reactive_ble #353: I could not get the device that was already connected, which caused me to be unable to disconnect the specified device

jahnli, 2021-09-02:
**Is your feature request related to a problem? (please describe)**

I could not get the device that was already connected, which caused me to be unable to disconnect the specified device  

**Describe the solution you'd like**

Similar to flutter blue. ConnectedDevices

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jahnli, 2021-09-02:
When I have successfully connected multiple devices, I should dispose the connected devices and close them.

jahnli, 2021-09-02:
When I have successfully connected multiple devices, how do I dispose the connected devices and close them?

Taym95, 2021-09-02:
Can't you implement this logic in your app? why you want it inside the library?

jahnli, 2021-09-02:
I did not find in the document how to obtain the connected device

jahnli, 2021-09-02:
I would like to get multiple connected devices and close them each time I dispose them, but I can't do it now

remonh87, 2021-09-02:
when you listen to a stream you will have a streamsubscription object. What you can do is add that collection to a Map and then cancel the specific subscription when you do not need it anymore.

jahnli, 2021-09-02:
Does that mean I need to add the map object successfully obtained from the connection to a list and then loop to call cancel()?

remonh87, 2021-09-03:
For example and if you use the deviceID as key then it is quite simple to disconnect specific devices

closed 2021-09-03 by remonh87

jahnli, 2021-09-03:
At present, there is a scenario: when I hot reboot on this page, the device cannot be disconnected and bluetooth remains connected

jahnli, 2021-09-03:
Is it possible to provide an API like Flutter_blue's connectedDevices