PhilipsHue/flutter_reactive_ble #355: App won't build on Mac M1

biklas7, 2021-09-06:
**Is your feature request related to a problem? (please describe)**
Can't build my app on a mac m1 with this package because of the used protobuf version.

**Describe the solution you'd like**
After doing some research I've found this issue and comment https://github.com/grpc/grpc-java/issues/7690#issuecomment-747196215 that had a solution for me.

**Additional context**
The build error without changing anything: `Could not find protoc-3.13.0-osx-aarch_64.exe`

remonh87, 2021-09-06:
I am able to build the example app with Mac m1 without any issues, also checked running from IDE and flutter run and both work without problems for me. I think you forgot to enable the rosetta emulation on your terminal (checkout this [article](https://medium.com/hackberry/setting-up-the-ios-development-environment-for-the-new-macbook-pro-m1-with-ease-16c1be7f997b)).

biklas7, 2021-09-06:
I think that was the problem. My bad, I'm new to Mac OS.

Now it works on the terminal but it's not working on VSCode. Do you have any tips for that as well?

closed 2021-09-06 by biklas7

remonh87, 2021-09-06:
No worries. About VSCode this should work out of the box however running on iOS is problematic sometimes.