PhilipsHue/flutter_reactive_ble #356: How to handle reconnects on device to far away or ble connection fails.

skast96, 2021-09-07:
Do you guys have a best practice example on how to handle auto reconnects on failures. Is the StreamSubscription var disposed after failure or trying to reconnect to the device itself do i have to call connectToAdvertisingDevice again and reinit the subscription?

remonh87, 2021-09-09:
true if you use `preScanAndConnect`  on Android we check if the device  is advertising in order to prevent the Android bleeding stack to wait for eternity to connect to the device.  What you can do for example is subscribe to: 
https://github.com/PhilipsHue/flutter_reactive_ble/blob/06f727777985403abc498cb5b246dbf137911119/packages/flutter_reactive_ble/lib/src/reactive_ble.dart#L62 

and then on an error or disconnect do a reconnect to the respective device

skast96, 2021-09-17:
Thank you for your answer. I did find another approach by only using "connectToDevice" and waiting for "onDone" if the connection state of the device is disconnected, I close the current subscription and retry to start a new subscription to "connectToDevice", which is working quite well.

closed 2021-10-04 by skast96