PhilipsHue/flutter_reactive_ble #357: Name is not coming of scanned devices

Abhishek01039, 2021-09-09:
**Describe the bug**
I am not able to get the name of the device which is scanned by this plugin.

**To Reproduce**
Steps to reproduce the behavior:
1. Scan for devices using `scanForDevices()`
2. get the stream of `DiscoveredDevice`
3. I am not able to get the name of the device (Only MAC address is coming)

**Expected behavior**
A clear and concise description of what you expected to happen.

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Asus zenfone pro M1
 - OS: [Android 9]
 - Package version: [4.0.0]

Abhishek01039, 2021-09-10:
@remonh87 
can you please look into it?
thanks

Taym95, 2021-09-10:
Can you please test it on nRF Connect and see if you are getting the name?

Abhishek01039, 2021-09-10:
I am getting the name of nordic devices

Thank you!

closed 2021-09-10 by Abhishek01039