PhilipsHue/flutter_reactive_ble #358: FATAL EXCEPTION: main

yuriboeira11tx, 2021-09-09:
**Describe the bug**
As soon as I cancel a connection subscription with a device, sometimes the app ends up closing. 

Subscription cancellation code:
```
try {
      print('Disconnecting to device: $deviceId');
      await deviceSubscription?.cancel();
      deviceSubscription = null;
} on Exception catch (e, _) {
 print("Error disconnecting from a device: $e");
}
```

Error and application closes: 
```
e.b.z.f: The exception could not be delivered to the consumer because it has already canceled/disposed the flow or the exception has nowhere to go to begin with. Further reading: https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling | d.c.a.l0.f: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 0 (GATT_SUCCESS)
```

Taym95, 2021-09-10:
I dont really understand your issue, can you please give steps to reproduce, we are implementing `cancelling connection subscription` in the example app and we see no error, I am afraid the issue is on your implementation not in the library.

yuriboeira11tx, 2021-09-10:
I just need to disconnect from the device. In the example application there is this subscription cancellation code and it works. It works in my app, but it closes.

yuriboeira11tx, 2021-09-10:
This is happening when the device disconnects from the app. Although it does not return an error, it is closing the application.

remonh87, 2021-09-11:
@yuriboeira11tx  you can fix this by following the instructions on: https://github.com/PhilipsHue/flutter_reactive_ble#how-to-handle-the-ble-undeliverable-exception

closed 2021-09-11 by remonh87

yuriboeira11tx, 2021-09-13:
> @yuriboeira11tx you can fix this by following the instructions on: https://github.com/PhilipsHue/flutter_reactive_ble#how-to-handle-the-ble-undeliverable-exception

Okay, I added these implementations and now the app doesn't compile. Any suggestion?

yuriboeira11tx, 2021-09-13:
In my project the Kotlin MainActivity exists, but not in Java.

Taym95, 2021-09-13:
Document about `FlutterApplication` you need to add that code in `onCreate`, please also document how to add `an application class in flutter` and how to add it to manifest

yuriboeira11tx, 2021-09-13:
Thanks for the explanation, but I found the solution right here. In a closed issue #345. To solve this problem, in any project it will be necessary to make these implementations. Will the ```flutter_reactive_ble``` library itself ever have this power?

remonh87, 2021-09-13:
@yuriboeira11tx  we will not integrate it in the library since it will suppress errors. Some consumer may consider handling things differently than the suggested workaround.

yuriboeira11tx, 2021-09-13:
I understand, but anyway. Thanks for the help, I really like this lib and its support.

obiwan007, 2023-03-21:
Would be great if you can at least not let the app crash as a default if no override is given?
If other consumers want to handle it differently they might be able to just override the default errorhandler?
Now everyone has a crashing app...