PhilipsHue/flutter_reactive_ble #360: characteristic not writeable on iOS

JimTompkins, 2021-09-13:
**Describe the bug**
Developing an app so far on Android, and now testing on iOS.  Getting a "characteristic is not writeable" error on iOS for something that works on Android.  It is able to scan and connect ok on iOS but throws an exception when writing.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to custom device using connectToDevice with just id and timeout i.e. no services/chars
2. wait for connection state connected
3. call QualifiedCharacteristic
4. write to the characteristic
5. get this error message:
2021-09-13 10:29:49.858099-0300 Runner[11498:736828] [VERBOSE-2:ui_dart_state.cc(199)] Unhandled Exception: Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "The characteristic 0000FFF6-0000-1000-8000-00805F9B34FB of the service 0000FFF0-0000-1000-8000-00805F9B34FB of the peripheral 2D2F69D2-A2DA-4648-237B-59FD642DF052 is not writable")

**Expected behaviour**
The write should work since it does on Android.

**Smartphone / tablet**
 - Device: iPhone6
 - OS: iOS 12.5.4
 - Package version: latest

**Peripheral device**
- Vendor, model: custom
- Does it run a custom firmware / software: yes

**Additional context**
On iOS do I have to explicitly request service and char discovery?  How do I wait for them to complete before writing, etc.?

Thanks!

remonh87, 2021-09-13:
@JimTompkins  can you check with a bluetooth explorer like nrf connect if writing on iOS works?

JimTompkins, 2021-09-13:
Yes, I can write that char in nRF connect on iOS.

Thanks,

Jim

JimTompkins, 2021-09-22:
I changed the writeCharacteristicWithoutResponse to one WithResponse and it works!  Not sure why, but I'll take it!

closed 2021-09-22 by JimTompkins

Mir1001, 2022-09-05:
@JimTompkins answer helped. The problem is just on the iOS device. Please open the issue.

gonzalolopez-tc, 2022-09-30:
@JimTompkins I'm facing this issue, any update about this?

JimTompkins, 2022-10-01:
Did you try changing the characteristic to "With Response" as noted above?

I have since changed from flutter_reactive_ble to flutter_blue_plus.  flutter_blue_plus is a fork from flutter_blue which is no longer maintained.

Hope this helps.

Mir1001, 2022-11-07:
> Did you try changing the characteristic to "With Response" as noted above?
> 
> I have since changed from flutter_reactive_ble to flutter_blue_plus. flutter_blue_plus is a fork from flutter_blue which is no longer maintained.
> 
> Hope this helps.

"With Response works", but still, it is a bug.

mashedbymachines, 2023-01-03:
I also have this exact issue when trying to write without response from an iOS device to our ESP32 chip with the bluedroid stack. However it is not convenient for me to change to withResponse as we need to maximize transfer speed when transfering a firmware blob and with response significantly slows things down. Has there been any solution to this?

JimTompkins, 2023-01-03:
My solution was to change to flutter_blue, which has since been superseded by flutter_blue_plus.

mashedbymachines, 2023-01-03:
> My solution was to change to flutter_blue, which has since been superseded by flutter_blue_plus.

Okay and that library did let you write without response, with no chance to the peripheral device code ? Interesting

I'm wondering if this really is a bug in the library or if it just more nitpicky strict about how the GATT characteristics needs to  be advertised from the device, and we are doing something slightly wrong with what we advertise in the GATT, that the Android code in this library, and iOS in the other library is just ignoring.

mashedbymachines, 2023-01-05:
Update for people still interested. It seems like this is NOT a bug in the library, but rather that iOS is more picky about the advertised permissions. 

We used this permission for our GATTs on a ESP32 chip: ESP_GATT_CHAR_PROP_BIT_WRITE, and that worked on Android and PC when writing with or without response. However for iOS we needed to also advertise this flag ESP_GATT_CHAR_PROP_BIT_WRITE_NR , so essentially ESP_GATT_CHAR_PROP_BIT_WRITE | ESP_GATT_CHAR_PROP_BIT_WRITE_NR.

Now it works on iOS too :)