PhilipsHue/flutter_reactive_ble #361: Android disconnecting causes crash with (GATT_SUCCESS), but not consistently

farr64, 2021-09-14:
**Describe the bug**
Android disconnecting causes crash with: 
`Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 0 (GATT_SUCCESS)`
but not consistently. Most of the time, Android disconnects perfectly.

(iOS does **not** exhibit this behavior.)

**To Reproduce**
This doesn't happen all the time and I haven't been able to detect a pattern.
So, this is just one case that I happened to catch while debugging and keeping an eye on the log.

Steps to reproduce the behavior:
1. With Android, connect to one device using `connectToAdvertisingDevice()`
2. Do lots of good stuff with the connected device.
3. `.cancel()` the connection

**Expected behavior**
A normal disconnection.

Most of the time, Android disconnects normally. In this case, it failed and it produced the attached log.

**Peripheral device**
- Nordic nRF52
- Does it run a custom firmware / software: yes

I have been using the App (under both Android and iOS) with several peripheral devices powered by Nordic's nRF52 and everything works great. So, this particular crash is an exception.

Perhaps the log file will help you track down some fundamental issue somewhere.

**Additional context**
Please see attached crash log.

Thanks.

[210913 2354 Android crash.txt](https://github.com/PhilipsHue/flutter_reactive_ble/files/7159196/210913.2354.Android.crash.txt)

remonh87, 2021-09-14:
I checked the log and it looks like it is an undeliverable exception. You can fix it by following instructions mentioned here: https://github.com/PhilipsHue/flutter_reactive_ble#how-to-handle-the-ble-undeliverable-exception . Let me know if this fixes it.

yuriboeira11tx, 2021-09-14:
Look in a closed issue #345.

farr64, 2021-09-15:
Thanks @remonh87. That helped.

farr64, 2021-09-15:
Thanks for the cross reference @yuriboeira11tx

closed 2021-09-15 by farr64

vanlooverenkoen, 2023-01-19:
I am still receiving this error even with the `BLE undeliverable exception` implementation

ElHyperion, 2023-04-15:
> Look in a closed issue #345.

The app builds with FATAL EXCEPTION after applying this code