PhilipsHue/flutter_reactive_ble #363: iOS 15 build failing on even new projects.

omarkamal, 2021-09-16:
**Describe the bug**
I have an existing project which uses this library without any issue. However, I started getting these errors (too much to display here so here's a pasetbin link https://pastebin.com/QRa7dDP9). 
I thought it was because of the migration to flutter 2.5 from 1.24. However, I opened up a dummy project and merely included this library in the pubspec file and this started happening. The build fails with the errors in the pastebin
**To Reproduce**
Steps to reproduce the behavior:
1. Create a new project 
2. Include this package
3. Run on an iOS 15 device 

**Expected behavior**
- Build fails with these errors (https://pastebin.com/QRa7dDP9)

**Smartphone / tablet**
 - Device: iPhone 11/iPhone 12 Pro Max Simulator
 - OS: iOS 15
 - Package version: 4.0.0
 - Xcode Version - Xcode 13 Beta 5 (to ensure it builds for iOS 15)

remonh87, 2021-09-17:
thanks I will have a look into it this weekend.

Zsmerritt, 2021-09-20:
Im also experiencing this issue. Any updates?

omarkamal, 2021-09-21:
@Zsmerritt there's a pull request (#365) pending which solves this

LEMUSADR000, 2021-09-21:
I seem to be getting this on all iOS >14.8. Thanks for getting to this so fast!

germancvdev, 2021-09-22:
Same problem since I update Xcode 13.0

michael-golfi, 2021-09-22:
Our iOS builds are failing as well. I tried using a git dependency pointing to #365 and it still seems to be failing. I'm not sure if it's just my machine. I cleaned the pub cache and ran flutter clean. Is it working well on your end?

xvrh, 2021-09-22:
It works for me with a `dependency_overrides` like that:

```yaml
dependency_overrides:
  flutter_reactive_ble:
    git:
      url: https://github.com/PhilipsHue/flutter_reactive_ble.git
      ref: 55604798e370ac7d8693f84778214cae0ea2ce47
      path: packages/flutter_reactive_ble
  reactive_ble_mobile:
    git:
      url: https://github.com/PhilipsHue/flutter_reactive_ble.git
      ref: 55604798e370ac7d8693f84778214cae0ea2ce47
      path: packages/reactive_ble_mobile
  reactive_ble_platform_interface:
    git:
      url: https://github.com/PhilipsHue/flutter_reactive_ble.git
      ref: 55604798e370ac7d8693f84778214cae0ea2ce47
      path: packages/reactive_ble_platform_interface
```

LEMUSADR000, 2021-09-22:
@xvrh thanks a ton, I needed to pull that branch and then fork off of it so that I could implement a feature that I wasn't able to get pulled in all the way before this update. You really saved my bacon!

remonh87, 2021-09-23:
fixed with the release of version `4.0.1`

closed 2021-09-23 by remonh87