PhilipsHue/flutter_reactive_ble #364: How to control how fast/frequent the DeviceConnectionState updates?

vontdeux, 2021-09-17:
Right now I am using the example app as a base of my app.

and by using the StreamProvider<ConnectionStateUpdate> and Consumer<ConnectionStateUpdate>, I managed to listen to the connection changes, but when I test to power off my Bluetooth Module without disconnecting it from app, it would take around 15-20 seconds for the app to get the ConnectionStateUpdate values.

is there a way to increase the frequency of this?

remonh87, 2021-09-17:
I think a lot depends on the peripheral vs phone. I noticed my updates on pixel come in around 5 seconds and on Samsung it takes 6-7 seconds. I think 20 seconds is too long so I would test on different phones (or using our example app) if you see more difference. On library level there is not much we can do, we propagate the update directly once we get it.

closed 2021-09-17 by remonh87

vontdeux, 2021-09-17:
hmm yes, after reading this and retesting it, on iphone8 it is immediately updated.