PhilipsHue/flutter_reactive_ble #366: Issue with building on Xcode12: Swift compiler error, Flutter/flutter.h not found

pierrot106, 2021-09-19:
I keep getting errors, and i've tried the following:

Android studio
- Clean the project
- Delete Podfile, Podfile.lock, pubspec.lock
- Delete DerivedData
- Delete Pods, .flutterplugin, .packages folders
- Rerun flutter package get
- Delete the Runner.xcodeproj/project.xcworkspace
- Delete ios folder & run flutter create
- pub upgrade

Xcode:
- swift suggested "swift 5 conversion" for reactive_ble_mobile frameword but wasn't unable to convert

Here's my Podfile and pubspec:
[code.pdf](https://github.com/PhilipsHue/flutter_reactive_ble/files/7190900/code.pdf)

Error results:
[error.pdf](https://github.com/PhilipsHue/flutter_reactive_ble/files/7190903/error.pdf)
[image]

remonh87, 2021-09-19:
This has nothing to do with our library. I am able to build the project with Xcode 12.5 without issues.  I would check https://github.com/flutter/flutter/issues/16036 .

closed 2021-09-19 by remonh87

pierrot106, 2021-09-20:
would you be able to guide me through how you set it up for xcode? Im able to run it on android studio, and i could run the app without flutter_reactive_ble. I've been trying for the past week with no avail.

pierrot106, 2021-09-20:
Is there a specific version of cocoapods or flutter that needs to be used, and is it required/ were you able to convert into swift 5?

pierrot106, 2021-09-20:
> This has nothing to do with our library. I am able to build the project with Xcode 12.5 without issues. I would check [flutter/flutter#16036](https://github.com/flutter/flutter/issues/16036) .
Thank for the suggestion I've tried following along flutter/flutter#16036, but i still can't resolve it.

I am able to run the app fine without just firebase plugins, only after i added flutter_reactive_ble plugin does it crash. I'm still quite new to flutter so I would appreciate if you can guide me through the setup, because it keeps telling me that it can't build the obj-c module of 'reactive_ble_mobile'. And says i need to convert it to swift 5.

I also tried setting up a new app with just the flutter_reactive_ble plugin, making sure that the swift code is selected and im still getting the same error.

Error message:
 <unknown>:0: error: could not build Objective-C module 'reactive_ble_mobile'
    Protobuf-gzfswgfoilxpnfdcvqbslkphozye
    Command CompileSwift failed with a nonzero exit code

ish2028, 2021-09-21:
I am getting build issues as well.  When running `flutter build ios` I get the following error.

```
.../ios/Runner/GeneratedPluginRegistrant.m:12:9: fatal error: module 'reactive_ble_mobile' not found
@import reactive_ble_mobile;
~~~~~~~^~~~~~~~~~~~~~~~~~~
1 error generated.
```

remonh87, 2021-09-21:
can you check if you can build the example app? also I would need a repro scenario because for me it is hard to triage this issue.

ish2028, 2021-09-25:
I was able to build the example project.  I will try to dig a little more to see what I can find.

ish2028, 2021-10-31:
@R-Emagina, I haven't had a chance to find a fix yet.  Hopefully this week I should be to work on it.  I will post an update on anything I find.

ish2028, 2021-11-02:
This error didn't appear again after I cleaned the project and configured the iOS deployment target in Xcode.