PhilipsHue/flutter_reactive_ble #367: No Device found with scanForDevices(withServices: serviceIds) method.

DhavalRKansara, 2021-09-22:
**Describe the bug**
I am trying to connect with the bluetooth of another android phone from my android phone, but by calling the `scanForDevices(withServices: serviceIds)` I do not found any device data in this(even after I am having two android phone and one wireless bluetooth on). Also I have one Hybrid smart watch and I am getting the device data for the particular watch but my use-case is to connect two android phone using bluetooth and implementation chat functionality between them.

**To Reproduce**
Steps to reproduce the behaviour:
1. Run the example given in the same package.
2.Start scanning for the devices.

**Expected behaviour**
`scanForDevices(withServices: serviceIds)` method should return other mobile devices and bluetooth device data as well.

- Note: I am passing empty list of uuid in the `scanForDevices` function.

SmartPhone in which app is running:
 - Device: Vivo V9
 - OS: Android 9
 - Package version: 4.0.0

Other phones with I want to connect:

1. Nokia 8.1 - (Android 10)
2. Lenovo K5 Note - (Android 6)

remonh87, 2021-09-22:
The problem is that ble works following:  one device acts as the `Central` which initiates the connection requests and the other device is the `peripheral` that is advertising. When scanning you will find only devices that are advertising and I do not think your device is doing this.  You need to write some code that makes sure this is happening. Our library does not facilitate for this use case but I found some useful blogpost that can push you in the right direction: https://code.tutsplus.com/tutorials/how-to-advertise-android-as-a-bluetooth-le-peripheral--cms-25426

closed 2021-09-22 by remonh87

DhavalRKansara, 2021-09-24:
@remonh87  Thanks for the quick reply.

I have gone through blog you have shared above and using the same I am able to advertise bluetooth from my second device.

At the same time it reflect in my [flutter_reactive_example](https://github.com/PhilipsHue/flutter_reactive_ble/tree/master/example) which is same as available in the repository. 

Function to establish connection with nearby bluetooth:
```
  Future<void> connect(String deviceId) async {
    _logMessage('Start connecting to $deviceId');
    _connection = _ble.connectToDevice(id: deviceId).listen(
      (update) {
        _logMessage(
            'ConnectionState for device $deviceId : ${update.connectionState}');
        _deviceConnectionController.add(update);
      },
      onError: (Object e) =>
          _logMessage('Connecting to device $deviceId resulted in error $e'),
    );
  }
```
After calling connectToDevice it does not sending any response(neither success not failure). If I set `connectionTimeout: const Duration(seconds: 10)` it gives **disconnect** status as update.

Note: I am using Nokia 8.1(Android 10) as second device to advertise BLE and using application in Vivo V9(Android 9) to connect with other BLE.