PhilipsHue/flutter_reactive_ble #368: Changing the connection interval

farr64, 2021-09-22:
**Is your feature request related to a problem? (please describe)**
For performance reasons, it would be useful to change the connection interval to optimize throughput depending on packet size.

**Describe the solution you'd like**
Some way to specify a specific connection interval.

**Describe alternatives you've considered**
None so far. Is there a way to specify a specific connection interval within your Library?

Thanks.

remonh87, 2021-09-22:
Based on your description I think you do not need a connection interval but negotiate an MTU size. See: https://github.com/PhilipsHue/flutter_reactive_ble/blob/06f727777985403abc498cb5b246dbf137911119/packages/flutter_reactive_ble/lib/src/reactive_ble.dart#L202

If this is not what you are looking for can you give me some more info?

remonh87, 2021-09-23:
Ah ok there is no way to change the interval but you can play a bit with the connection priority in android by using: 

https://github.com/PhilipsHue/flutter_reactive_ble/blob/06f727777985403abc498cb5b246dbf137911119/packages/flutter_reactive_ble/lib/src/reactive_ble.dart#L210

For example if you use `ConnectionPriority. highPerformance ` you will request a connection with low latency and high performance. Of course it is up to the specific vendor and probably also your peripheral to negotiate a proper interval.

closed 2021-09-23 by remonh87

farr64, 2021-09-23:
Great to know. Thanks for your wise guidance, @remonh87