PhilipsHue/flutter_reactive_ble #370: Android 12 permission system

juliansteenbakker, 2021-09-23:
**Describe the bug**
With Android 12 new permissions are introduced which are required to utilize bluetooth functionality.

**To Reproduce**
Steps to reproduce the behavior:
Right now, projects which target API 31 are failing because the permission android.permission.BLUETOOTH_SCAN is not set.

**Expected behavior**
Apply the new permissions as described here: https://developer.android.com/about/versions/12/features/bluetooth-permissions

The page also describes how to use `android:maxSdkVersion="30"` on the existing permissions to keep backwards compatibility with older SDK.

**Smartphone / tablet**
All android 12 devices.

PieterAelse, 2021-09-23:
Hi Julian, we already have issue https://github.com/PhilipsHue/flutter_reactive_ble/issues/273 and it's linked to this issue in Polidea library: https://github.com/Polidea/RxAndroidBle/issues/752

juliansteenbakker, 2021-09-23:
My bad, i didn't saw it right away. Maybe it's a good idea to pin the issue since it's blocking Android 12 apps and i'm sure more people will run into this.

closed 2021-09-23 by juliansteenbakker

PieterAelse, 2021-09-23:
@juliansteenbakker Good suggestion! I've pinned the issue 👍