PhilipsHue/flutter_reactive_ble #372: Latest 4.0.1 release seems to not work on versions of Xcode < 13.0

LEMUSADR000, 2021-09-23:
**Describe the bug**
Backwards compatibility with <Xcode 13.0 seems to be broken in the 4.0.1 release when attempting to build & archive project

**To Reproduce**
Steps to reproduce the behavior:
1. Clone flutter_reactive_ble project fresh and depend on 4.0.1
2. Attempt to run project using build tools, below is the command attached to my fastlane gym build which was failing for my project. I imagine the relevant bits may be plucked out.

`set -o pipefail && xcodebuild -workspace Runner.xcworkspace -scheme Runner -configuration Release -destination 'generic/platform=iOS' -archivePath /Users/tech/Library/Developer/Xcode/Archives/2021-09-23/Runner\ 2021-09-23\ 13.24.27.xcarchive -allowProvisioningUpdates clean archive | tee /Users/tech/Library/Logs/gym/Runner-Runner.log | xcpretty`

**Expected behavior**
Since it didn't seem to be stated anywhere, I figured this would be backwards compatible and would be archivable successfully on any version of Xcode, but this project seems to have issues with backwards compatibility as of 4.0.1. If this is expected and or a known issue, we may want to add this note in for any users who aren't yet running Xcode 13 so they may hold off on updating the project to the latest version.

remonh87, 2021-09-24:
yes you are right you can fix it by using dependency overrides as I stated in: #373 .  I will add it to the readme.

closed 2021-09-24 by remonh87

bretmh, 2021-09-24:
After trying your fix the problem persists. Might require some cleanup before using this fix? Any suggestions?

I've removed all reactive_ble_x dependencies and added `reactive_ble_mobile: '4.0.0'` in the dependency overrides. Any packages that are also dependent on reactive_ble have also been updated with the same dependency override.

bretmh, 2021-09-24:
For a reproducible result using the example app. I have updated the `pubspec.yaml` as the following.

Xcode: Version 12.5.1 (12E507)

```
name: flutter_reactive_ble_example
description: Demonstrates how to use the flutter_reactive_ble plugin.
version: 3.1.1+1
publish_to: 'none'

environment:
  sdk: '>=2.12.0 <3.0.0'
  flutter: ">=1.10.0"

dependencies:
  flutter:
    sdk: flutter
  flutter_lints: ^1.0.3
  # flutter_reactive_ble:
  #   path: ../packages/flutter_reactive_ble
  functional_data: ^1.0.0
  intl: ^0.17.0

  provider: ^6.0.0

dev_dependencies:
  build_runner: ^2.1.2
  dependency_validator: ^3.1.0
  flutter_test:
    sdk: flutter
  functional_data_generator: ^1.1.2
  reactive_ble_mobile: '4.0.0'

flutter:
  uses-material-design: true
```

Output from a `flutter clean && flutter build ios`
```
flutter clean && flutter build ios
Cleaning Xcode workspace...                                      2,756ms
Deleting build...                                                    9ms
Deleting .dart_tool...                                               0ms
Deleting .packages...                                                0ms
Deleting Generated.xcconfig...                                       0ms
Deleting flutter_export_environment.sh...                            0ms
Deleting Flutter.podspec...                                          0ms
Deleting .flutter-plugins-dependencies...                            0ms
Deleting .flutter-plugins...                                         0ms
Running "flutter pub get" in example...                            776ms
Building com.signify.huereactivebleExample for device (ios-release)...
Found saved certificate choice "Apple Development: Joshua Diaz (44Q68PRHT9)". To
clear, use "flutter config".
Signing iOS app for device deployment using developer identity: "Apple
Development: Joshua Diaz (44Q68PRHT9)"
Running pod install...                                           1,523ms
Running Xcode build...                                                  
Xcode build done.                                           52.2s
Failed to build iOS app
Error output from Xcode build:
↳
    2021-09-24 09:35:42.324 xcodebuild[47557:7880252]
    CFURLRequestSetHTTPCookieStorageAcceptPolicy_block_invoke: no longer
    implemented and should not be called
    2021-09-24 09:35:42.884 xcodebuild[47557:7880252]  DVTPortal: Service
    '<DVTPortalViewDeveloperService: 0x7fb9211e2950; action='viewDeveloper'>'
    encountered an unexpected result code from the portal ('1100')
    2021-09-24 09:35:42.885 xcodebuild[47557:7880252]  DVTPortal: Error:
    Error Domain=DVTPortalServiceErrorDomain Code=1100 "Your session has
    expired. Please log in." UserInfo={payload={
[... 677 lines cut ...]
    <asynchronous suspension>
    #3      FrontendCompiler.compile
    (package:frontend_server/frontend_server.dart:562:9)
    <asynchronous suspension>
    #4      starter (package:flutter_frontend_server/server.dart:180:12)
    <asynchronous suspension>
    #5      main
    (file:///opt/s/w/ir/cache/builder/src/flutter/flutter_frontend_server/bin/st
    arter.dart:13:24)
    <asynchronous suspension>

    Command PhaseScriptExecution failed with a nonzero exit code
    note: Using new build system
    note: Building targets in parallel
    note: Planning build
    note: Analyzing workspace
    note: Constructing build description
    note: Build preparation complete
    /Users/bret/work/flutter_reactive_ble/example/ios/Pods/Pods.xcodeproj:
    warning: The iOS deployment target 'IPHONEOS_DEPLOYMENT_TARGET' is set to
    8.0, but the range of supported deployment target versions is 9.0 to
    14.5.99. (in target 'Flutter' from project 'Pods')

Encountered error while building for device.
```

remonh87, 2021-09-24:
this doesn't work because you still need to put `flutter_reactive_ble` in the path. 

```yaml
ependencies:
  flutter:
    sdk: flutter
  flutter_lints: ^1.0.3
  flutter_reactive_ble: ^4.0.1 #can also be 4.0.0
  functional_data: ^1.0.0
  intl: ^0.17.0

  provider: ^6.0.0

dev_dependencies:
  build_runner: ^2.1.2
  dependency_validator: ^3.1.0
  flutter_test:
    sdk: flutter
  functional_data_generator: ^1.1.2

flutter:
  uses-material-design: true

dependency_overrides:
  reactive_ble_mobile: 4.0.0
```

bretmh, 2021-09-24:
> this doesn't work because you still need to put `flutter_reactive_ble` in the path. I could compile the example app with this:
> 
> ```yaml
> ependencies:
>   flutter:
>     sdk: flutter
>   flutter_lints: ^1.0.3
>   flutter_reactive_ble: ^4.0.1 #can also be 4.0.0
>   functional_data: ^1.0.0
>   intl: ^0.17.0
> 
>   provider: ^6.0.0
> 
> 
> dev_dependencies:
>   build_runner: ^2.1.2
>   dependency_validator: ^3.1.0
>   flutter_test:
>     sdk: flutter
>   functional_data_generator: ^1.1.2
> 
> flutter:
>   uses-material-design: true
> 
> dependency_overrides:
>   reactive_ble_mobile: 4.0.0
> ```

Dependency `flutter_reactive_ble: ^4.0.1` had to be `flutter_reactive_ble: ^4.0.0` for `flutter pub get` to succeed. My current Flutter SDK version is `2.2.3` and it had conflicts with my version. 

Still having issues with this.

anthonyhunter, 2021-09-25:
> #373

Confirmed fix with Flutter 2.5.1 and Xcode tools 12.5.1