PhilipsHue/flutter_reactive_ble #373: Unavailable Features + Syntax Issues

bretmh, 2021-09-24:
**Describe the bug**
When running `flutter build ios` I receive errors for reactive_ble_mobile.

**To Reproduce**
Steps to reproduce the behavior:
1. Xcode 12.5.1
2. flutter_reactive_ble: ^4.0.0
3. Make sure my projects ios/Podfile's `platform :ios` is > 10
4. flutter clean && flutter build ios

**Expected behavior**
For flutter build ios to complete with no critical errors.

**Peripheral device**
- n/a

**Additional context**
I have deleted all of my archived data, flutter cleaned, redownloaded packages, set the project's minimum deployment to 10.0 in the ios/Podfile and within settings of Xcode. The builds were working just fine a couple of hours ago.

remonh87, 2021-09-24:
we launched a new version with support for Xcode 15 the annoying things is that the pub dev eco system does not allow us to pin versions. What happens is that with new builds for flutter_reactive_ble it gets the transitive dependency with version `4.0.1` which contains fixes for the new Xcode.  

You can fix it by using a dependency_override in your app:

```yaml
dependency_overrides:
  reactive_ble_mobile: '4.0.0'
```

closed 2021-09-24 by remonh87

ride4sun, 2022-01-10:
> I will add this to the readme

was this added to the readme? Can not find it... I still get this error