PhilipsHue/flutter_reactive_ble #378: ConnectionTimeout is set. It doesn't seem to work

jahnli, 2021-09-26:
[image]

When I set up a second overtime, it will not be disconnected or return an error, bluetooth status has also been a DeviceConnectionState. Connecting

jahnli, 2021-09-26:
How do I catch mistakes, how to terminate DeviceConnectionState. Connecting

remonh87, 2021-09-27:
we will terminate the stream when the device gets disconnected so you probably will not get the update. However you can use the https://github.com/PhilipsHue/flutter_reactive_ble/blob/876de0df13410a93a9034f39e65dce7833bd7126/packages/flutter_reactive_ble/lib/src/reactive_ble.dart#L62  in order to check if the device is disconnect or use `onDone` function in the stream.listen .

jahnli, 2021-09-28:
> 当设备断开连接时，我们将终止流，因此您可能无法获得更新。但是，您可以使用
> 
> https://github.com/PhilipsHue/flutter_reactive_ble/blob/876de0df13410a93a9034f39e65dce7833bd7126/packages/flutter_reactive_ble/lib/src/reactive_ble.dart#L62
> 
> 为了检查设备是否断开连接或使用`onDone`stream.listen 中的功能。

I can get a DeviceConnectionState by checking the flow, but I can't disconnect the device from that state

remonh87, 2021-09-28:
that is what I meant you can use the operation that I suggested and you will get a disconnected event. The way this library works is that we disconnect a device once the stream is cancelled.

closed 2021-10-04 by remonh87