PhilipsHue/flutter_reactive_ble #385: iOS: Warning! No event channel set up to report a connection update, notifications not working

JimTompkins, 2021-09-27:
**Describe the bug**
I get a "Warning! No event channel set up to report a connection update" when trying to connect on iOS.  The warning is not present on Android with similar code (on Android I use connectToAdvertisingDevice and on iOS connectTo).

On Android, I can successfully subscribe to a notification.  On iOS, I do not receive any notifications.   I'm not sure if this warning is related to the notification issue or not.  I can successfully connect, read and write on iOS (although I had to change from writeWithoutResponse to writeWithResponse,  as others have suggested).

**To Reproduce**
Steps to reproduce the behavior:
1. try to connect with this code: [ble.dart](https://github.com/JimTompkins/happy_feet_app/blob/d9c497f47d0ca0a1a9cfd205edfe94427a28881a/lib/ble.dart)

**Expected behavior**
Should connect without the warning as it does on Android.

**Smartphone / tablet**
 - Device: iPhone6
 - OS: iOS13
 - Package version: 4.0.0

**Peripheral device**
- Vendor, model: custom
- Does it run a custom firmware / software: yes

**Additional context**
The missing notifications on iOS work on Android and also using nRF Connect.

remonh87, 2021-09-29:
hmm this looks like a bug (thanks for the elaborate example btw). Can you try to add a small delay (500ms) as a workaround? before subscribing to the notifications?  It feels like a race conditio.

JimTompkins, 2021-09-29:
Thanks for the feedback!

I initially tried subscribing immediately after the connection was established.  Now, I'm subscribing on a button press which is typically seconds after the connection is established in case there needed to be a delay after the connection is made for services to be discovered.  This didn't work either.

Do you think the "no event channel" warning is related to the fact that notifications are not being received, or a separate issue? 

I also had an issue ($360) where writeCharacteristicWithoutResponse didn't work on iOS so I changed it to WithResponse as a workaround.  Not sure if this could be related or not...

mwernsen, 2021-10-01:
I've also been observing the "Warning! No event channel set up to report a connection update" warning on ios. The warning appears as soon as the connectTo command is invoked. Connecting works fine, however the **_ConnectionStateUpdate.Connecting_** event is never added to the stream. This is ONLY the case for the first device I connect to after an app restart. Second, third nth device all work as expected. 

Notifications, reading and writing characteristics are not affected by this warning after a connection has been established and this behaviour is not present on android.

**Smartphone / tablet**

Device: iPhone7
OS: iOS14
Package version: 4.0.1

JimTompkins, 2021-10-01:
I checked and I only get this warning on the first connection after installing the app, the same as @mwernsen .  It does not occur on the 2nd connection.

I've also found that readCharacteristics are also not working in addition to notifications.  Both work in the Android version and on iOS using nRF Connect.  I have try and catch blocks around both reads and subscriptions but do not get any error messages.

JimTompkins, 2021-10-04:
I'm using Catalina OS so I'm limited to Xcode v12.4.  I saw that v4.0.1 is for Xcode 13 so I changed back to version 4.0.0 of flutter_reactive_ble but this didn't make any change.  I also tried using v4.0.1 with a dependency_override for reactive_ble_mobile to '4.0.0'.

The log messages on iOS are much more sparse than Android but I can't find a way to turn them on in Xcode.

Any suggestions for further digging?

Thanks!

JimTompkins, 2022-02-10:
Hi Gaelle, because of this issue I changed to flutter_blue_plus for the iOS version.  It's not my preferred solution to use different packages for Android and iOS!

tomstokes, 2022-02-15:
Digging deeper, the #406  commit that closed this issue was reverted in #440 . Looks like this issue should be re-opened.

dlewis2017, 2022-02-15:
> I actually mannage to read the characteristic now ! I realise that i needed to use 16 bit ID and NOT 128 bits ID (which worked in Android!), the read on 128 bit ID were causing infinite read with no answer but no error either. With 16 bit characteristic id (the same you got when you do discover characteristics), it worked fine :)

This seemed to work for me. For some reason, it gave the above unhelpful errors. It doesn't like things in the format '180f', such as when passing `Uuid.parse('180F')` to the service/characteristic value(s) in subscribing to a characteristic. Not sure if that's an issue with the parsing function or inherent to iOS. I haven't been able to test on android so can't confirm there.

GaelleJoubert, 2022-02-16:
I've tested it both with Android and IOS. 
For Android, I have to use the 128 bit ID, other it doesn't work, and for IOS I have to use the 16 bit ID. 
So I'm doing something like that :

`final SERVICE_DEVICE_INFO = Platform.isAndroid ? "0000180A-0000-1000-8000-00805F9B34FB" : "180a";`

And then later I call that string with Uuid.parse : `Uuid.parse(SERVICE_DEVICE_INFO )`

RikiZimbakov, 2022-06-24:
this thread is amazing! thank you to @GaelleJoubert for the insight! 

for reference I dont know if it has been updated since then but for me only 16bit works for both service and characteristic. For android and IOS it works being 16bit too so its perfect! 

Thank you all for your help!

chipweinberger, 2023-05-02:
I am also seeing this error message, and failing to stay connected for more than a ~0.5 seconds. 

https://github.com/PhilipsHue/flutter_reactive_ble/issues/728

Is the the 16-bit UUID relevant to me? I am not reading or writing a characteristic yet. Just trying to stay connected!

nissaba, 2023-09-26:
getting this error also. if I re-run / hot reload the app then I do get 1 update on the listen of the connection.