PhilipsHue/flutter_reactive_ble #388: Ble Connection Closes Immediately.

shobhit14gupta, 2021-09-28:
[image]

From the above image:
Wih the reference from your flutter_reactive_ble example, as soon as I call connectOnDevice method, the device doesn't wait for the callback from the peripheral device and close the connection even without any statusCode. 

Kindly help!

yuriboeira11tx, 2021-09-28:
Code?

shobhit14gupta, 2021-09-29:
Here is the Code:

` Future<void> connect(DiscoveredDevice device) async {
    if (await _bleAdvertise.isAdvertise()) _bleAdvertise.oneSecAdvertise();
    _deviceID = device.id;
    _connection = _ble
        .connectToDevice(
            id: device.id,
            connectionTimeout: Duration(milliseconds: ConfigTimeout.connect))
        .listen(
      (update) async {
        if (update.failure != null) {
          if (update.failure.code == ConnectionError.failedToConnect) {
            await disconnect();
            await Future.delayed(
              Duration(milliseconds: 500),
              () async => await connect(device),
            );
          }
        } 
        _deviceConnectionController.add(update);
      },
      onError: (Object e) =>
          debugPrint('Connecting to device $_deviceID resulted in error $e'),
    );
  }

  Future<void> disconnect() async {
    try {
      debugPrint('disconnecting to device: $_deviceID');
      if (Platform.isAndroid) {
        await _clearGattCache(id);
      }
      await _connection.cancel();
    } on Exception catch (e, _) {
      debugPrint("Error disconnecting from a device: $e");
    } finally {
      // Since [_connection] subscription is terminated, the "disconnected" state cannot be received and propagated
      _deviceConnectionController.add(
        ConnectionStateUpdate(
          deviceId: "Unknown Device",
          connectionState: DeviceConnectionState.disconnected,
          failure: null,
        ),
      );
    }
  }`

shobhit14gupta, 2021-09-29:
Here is the code:

```
 Future<void> connect(DiscoveredDevice device) async {
    if (await _bleAdvertise.isAdvertise()) _bleAdvertise.oneSecAdvertise();
    _deviceID = device.id;
    _connection = _ble
        .connectToDevice(
            id: device.id,
            connectionTimeout: Duration(milliseconds: ConfigTimeout.connect))
        .listen(
      (update) async {
        if (update.failure != null) {
          if (update.failure.code == ConnectionError.failedToConnect) {
            await disconnect();
            await Future.delayed(
              Duration(milliseconds: 500),
              () async => await connect(device),
            );
          }
        } 
        _deviceConnectionController.add(update);
      },
      onError: (Object e) =>
          debugPrint('Connecting to device $_deviceID resulted in error $e'),
    );
  }

  Future<void> disconnect() async {
    try {
      debugPrint('disconnecting to device: $_deviceID');
      if (Platform.isAndroid) {
        await _clearGattCache(id);
      }
      await _connection.cancel();
    } on Exception catch (e, _) {
      debugPrint("Error disconnecting from a device: $e");
    } finally {
      // Since [_connection] subscription is terminated, the "disconnected" state cannot be received and propagated
      _deviceConnectionController.add(
        ConnectionStateUpdate(
          deviceId: "Unknown Device",
          connectionState: DeviceConnectionState.disconnected,
          failure: null,
        ),
      );
    }
  } 
```

shobhit14gupta, 2021-09-29:
> Code?

Here is the code:

```
 Future<void> connect(DiscoveredDevice device) async {
    if (await _bleAdvertise.isAdvertise()) _bleAdvertise.oneSecAdvertise();
    _deviceID = device.id;
    _connection = _ble
        .connectToDevice(
            id: device.id,
            connectionTimeout: Duration(milliseconds: ConfigTimeout.connect))
        .listen(
      (update) async {
        if (update.failure != null) {
          if (update.failure.code == ConnectionError.failedToConnect) {
            await disconnect();
            await Future.delayed(
              Duration(milliseconds: 500),
              () async => await connect(device),
            );
          }
        } 
        _deviceConnectionController.add(update);
      },
      onError: (Object e) =>
          debugPrint('Connecting to device $_deviceID resulted in error $e'),
    );
  }

  Future<void> disconnect() async {
    try {
      debugPrint('disconnecting to device: $_deviceID');
      if (Platform.isAndroid) {
        await _clearGattCache(id);
      }
      await _connection.cancel();
    } on Exception catch (e, _) {
      debugPrint("Error disconnecting from a device: $e");
    } finally {
      // Since [_connection] subscription is terminated, the "disconnected" state cannot be received and propagated
      _deviceConnectionController.add(
        ConnectionStateUpdate(
          deviceId: "Unknown Device",
          connectionState: DeviceConnectionState.disconnected,
          failure: null,
        ),
      );
    }
  } 
```

remonh87, 2021-09-29:
It looks like that on the android side the connection is immediately closed probably due to the fact the stream is terminated. I would double check  if your stream is for some reason cancelled. Can you also check on our example app if the same behaviour is happening?