PhilipsHue/flutter_reactive_ble #394: v4.0.1 contains nullability errors when integrated with Xcode 13.

vmammanalupo, 2021-10-01:
I know there were other reports of this using a version of Xcode < 13, but I have read those threads carefully and am at a bit of a loss to why the plugin contains these issues. Part of the mystery is that my other team members do experience any issues. We have aligned on tooling:

Xcode 13.0
Flutter 2.5.1 SDK
Dart 2.14.2
Cocoapods 1.11.2
flutter_reactive_ble 4.0.1

I have 
- completely nuked all caches
- cleared derived data
- cleaned the project 
- restarted the computer
- removed build folders 
- did a complete reinstall of dependencies for the flutter project. 

Android doesn't have any issues but iOS contains the prior reported code issues. Were you guys able to find out any other information from the closed threads? This seems like another component is problematic, as the proposed PR seems to be a fixed designed for users of older versions ( < 13)  of Xcode. 

**To Reproduce**
Steps to reproduce the behavior:
1. Integrate v4.0.1 of `flutter_reactive_ble`
2. Attempt to build for iOS
3. Observe build failures 
[image]

**Expected behavior**
No errors with the module

**Smartphone / tablet**
 - Device: iPhone XS Max Simulator
 - OS: iOS 15
 - Package version: 4.0.1

**Peripheral device**
- Vendor, model: Custom
- Does it run a custom firmware / software: yes

**Additional context**
This is more of a shot in the dark. Again, I do not doubt that this works for some people, but I was curious to be able to understand why it is not working as intended for me. Was just hoping for suggested recommendations on what else I can do here. Related threads [Issue 372](https://github.com/PhilipsHue/flutter_reactive_ble/issues/372), [Issue 373](https://github.com/PhilipsHue/flutter_reactive_ble/issues/373), [Issue 386](https://github.com/PhilipsHue/flutter_reactive_ble/issues/386) all seem to have issues with older versions of Xcode, but I am using Xcode 13.

remonh87, 2021-10-01:
@vmammanalupo  it looks like you have an old version of the `reactive_ble_mobile` library. Can you make sure you have set that version to `4.0.1` in your pub spec.lock ?

vmammanalupo, 2021-10-01:
@remonh87 Thanks so much for the reply. 

I did confirm that I do have `4.0.1` integrated in my `pubspec.yaml` and `pubspec.lock` .  I still deleted the pubspec.lock, re-performed a `flutter pub get` and a `pod install`. Same issue occurs when trying to build. 

[image]

Do you know if there is any versioning with the `reactive_ble_mobile` library? looks to be set to `0.0.1`.

vmammanalupo, 2021-10-01:
[image]

Here is an image of the `pubspec.yaml` as well. It is set to `4.0.1`. We avoid any `^` usage so CI doesn't have the potential to update plugins.

meamde, 2021-10-01:
i have a same error when build with xcode 12.4 and flutter 2.5...

jalpedersen, 2021-10-01:
Check that your iOS deployment target is >= 11

vmammanalupo, 2021-10-01:
> _i have a same error when build with xcode 12.4 and flutter 2.5..._
> 
> oh, my problem solved with dependcy_overrides 4.0.0.

Yeah the dependency override is a solution for users with versions of Xcode < 13. I am experiencing problems on Xcode 13.0 which isn't covered.

vmammanalupo, 2021-10-01:
@jalpedersen 

> Check that your iOS deployment target is >= 11

I have the ios version set to `13.0` for the main target and all pods

[image]

[image]

vmammanalupo, 2021-10-01:
I have tried playing around with the deployment target values, but I always make sure to `pod install` after to make sure it actually integrates the changes

vmammanalupo, 2021-10-01:
[image]

From the podfile.lock I confirmed with my team members that it _looks to be pulling in the same library commit_ as what is working for them

vmammanalupo, 2021-10-01:
Cocoapods seemed to be deceiving me on what it was integrating... I removed the library from the `pubspec.yaml` completely and ran a  `flutter pub get` and then a `pod install`.

Attempted to build the project knowing it would fail without the `flutter_reactive_ble` plugin.

Then performed `pod cache clean --all` and completely deleted the `podfile.lock`, `Pods` directory and the `build` directory in ios.

Did a `flutter clean` and readded the `flutter_reactive_ble 4.0.1` to the pubspec.yaml and `reintegrated` everything again.

Thanks for the help guys

closed 2021-10-01 by vmammanalupo