PhilipsHue/flutter_reactive_ble #396: Please add protobuf chages to get it build on Apple Silicon M1

y60yu1ii, 2021-10-01:
**Describe the bug**
There is an issue of protobuf that it does not get the correct artifact name, which cause the plugin does not build to Android on Apple silicon M1

**To Reproduce**
1. Add package to the project.
2. Build & Run on Android.
3. You would see 

`FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':reactive_ble_mobile:generateDebugProto'.
> Could not resolve all files for configuration ':reactive_ble_mobile:protobufToolsLocator_protoc'.
   > Could not find protoc-3.13.0-osx-aarch_64.exe (com.google.protobuf:protoc:3.13.0).
     Searched in the following locations:
         https://repo.maven.apache.org/maven2/com/google/protobuf/protoc/3.13.0/protoc-3.13.0-osx-aarch_64.exe

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 6s
Exception: Gradle task assembleDebug failed with exit code 1`

**Expected behavior**
it builds

**Smartphone / tablet**
 - Device:  Pixel2
 - OS: Android 10
 - Package version: 3.3.1+1, 4.0.0, 4.0.1

**Peripheral device**

**Additional context**
Please made the following changes to the gradle file.
`
protobuf {
    protoc {
        // a fix for m1
        if (osdetector.os == "osx") {
            artifact = 'com.google.protobuf:protoc:3.13.0:osx-x86_64'
        } else {
            artifact = 'com.google.protobuf:protoc:3.13.0'
        }
    }

    generateProtoTasks {
        all().each { task ->
            task.builtins {
                java {
                    option "lite"
                }
            }
        }
    }
}`

remonh87, 2021-10-01:
do you use rosetta emulation for your command terminal? I am able to build the app with ease on my Mac m1. I also do not know the side effects of integrating this for non m1 machines.

y60yu1ii, 2021-10-01:
yes, I have tried using the rosetta the result is the same. 

This issue is addressed here
https://github.com/grpc/grpc-java/issues/7690

I previously forking the 3.3.1+1 project and applied the changes to make it build.
However, directly import the github repo to pubspec.yaml doesn't seem working after federated plugin structure.

Thank you for the information, it is good to konw that there is another m1 is able to run the project.
I will see what I missed on my set up.

remonh87, 2021-10-11:
I will have a look into this issue the coming time. I finally could reproduce it on my m1.

y60yu1ii, 2021-10-13:
I found something new here.
https://github.com/protocolbuffers/protobuf/issues/8062

The issue seems to be there is no file called such name.

davormaricdev, 2021-10-15:
Bumping artifact to 3.18.1 might resolve the issue as that package contains an reference to `protoc-xxx-osx-aarch_64.exe`
Source: https://repo1.maven.org/maven2/com/google/protobuf/protoc/3.18.1/

y60yu1ii, 2021-11-18:
It works like a charm.