PhilipsHue/flutter_reactive_ble #397: Location Permission missing (code 3)

meridorkobi, 2021-10-02:
I'm trying to scan in the background and I have this error
I also tried with requireLocationServicesEnabled:true

**Code**
```
      flutterReactiveBle.scanForDevices(withServices: [],requireLocationServicesEnabled:false).listen((device)  {
          //code for handling results
          print("service - blue - " + device.name.toString());
          }, onError: (error) {
            print('service - blue - error!' + error.toString());
          });
```
**plugins**
lutter_reactive_ble - 4.0.1
flutter_background_service - 0.1.5

**full error**
 GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Location Permission missing (code 3)")

remonh87, 2021-10-02:
you need to enable location permissions. See https://github.com/PhilipsHue/flutter_reactive_ble#which-permissions-are-needed

closed 2021-10-02 by remonh87

meridorkobi, 2021-10-02:
I did it
    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />

meridorkobi, 2021-10-02:
> you need to enable location permissions. See https://github.com/PhilipsHue/flutter_reactive_ble#which-permissions-are-needed

Can you reopen please

remonh87, 2021-10-02:
but also you need to make sure your permission is granted.

meridorkobi, 2021-10-02:
> but also you need to make sure your permission is granted.

Can I scan without the location.
I don't need the location

EdoardoVignati, 2022-02-18:
I fixed requesting the permission:
`await Permission.location.request();`