PhilipsHue/flutter_reactive_ble #398: Broadcast as a beacon

faheem4797, 2021-10-04:
Does this package allows us to broadcast our device as a beacon? If yes then can you please point me towards the method/property. 
Or does this package just allows us to scan for devices and i have to use something else to first broadcast as a beacon?

faheem4797, 2021-10-04:
So I did broadcast as a beacon using 'beacon_broadcast'. [https://pub.dev/packages/beacon_broadcast](url) 
[image]

And I have checked the results in Beacon Simulator (an app on play store). Which is working fine and UUID is visible etc.
[image]

The issue is that when I scan the same beacon through 'flutter_reactive_ble'. The result doesnt show any service data.
[image]

I am also calling scanfordevices with the uuid but not getting the desired result. I want serviceData to be shown here.
[image]

Can you please tell me what i am doing wrong here? or point towards the solution. Will be much thankful.

Both the devices are Android 9 (API 28).

remonh87, 2021-10-04:
can you check with a bluetooth explorer like nrf connect if the device has service data? The good thing is that the advertisement packet is there but for some reason the service data is not filled are you sure that the device has service data?

closed 2021-10-16 by remonh87

matiasleyba, 2022-04-24:
@faheem4797 can you explain the solution ? I have a simliar issue... I cant detect the device

faheem4797, 2022-04-25:
@matiasleyba i think i dropped the whole service data idea and just sent some data through manufacturerdata and used some logic to compare after scanning  (i was making a school project so not recommended for actual industry level solution).

 Also instead of "beacon_broadcast" package (https://pub.dev/packages/beacon_broadcast) used the flutter_ble_peripheral (https://pub.dev/packages/flutter_ble_peripheral) (not recommended if you are making ios app aswell)

matiasleyba, 2022-04-25:
@faheem4797 thanks for response, I resolved the issue, it was bad code, i forgot to manage the beacon layout correctly