PhilipsHue/flutter_reactive_ble #400: Not able to connect with scanned device

DhavalRKansara, 2021-10-05:
I have gone through blog you have shared in [#367](https://github.com/PhilipsHue/flutter_reactive_ble/issues/367)  and using the same I am able to advertise bluetooth from my second device.

At the same time it reflect in my [flutter_reactive_example](https://github.com/PhilipsHue/flutter_reactive_ble/tree/master/example) which is same as available in the repository. 

Function to establish connection with nearby bluetooth:
```
  Future<void> connect(String deviceId) async {
    _logMessage('Start connecting to $deviceId');
    _connection = _ble.connectToDevice(id: deviceId).listen(
      (update) {
        _logMessage(
            'ConnectionState for device $deviceId : ${update.connectionState}');
        _deviceConnectionController.add(update);
      },
      onError: (Object e) =>
          _logMessage('Connecting to device $deviceId resulted in error $e'),
    );
  }
```
After calling connectToDevice it does not sending any response(neither success not failure). If I set `connectionTimeout: const Duration(seconds: 10)` it gives **disconnect** status as update.

May be it's something related to issues `[#354](https://github.com/PhilipsHue/flutter_reactive_ble/issues/354)`, and so on I tried to pass connectionTimeout as well, but still not able to connect.
 
**Note:** I am using Nokia 8.1(Android 10) as second device to advertise BLE and using application in Vivo V9(Android 9) to connect with other BLE.

remonh87, 2021-10-05:
Can you also try to connect to the device using our example app?  And also can you provide some logical output?

DhavalRKansara, 2021-10-06:
@remonh87 I have cloned your completed repository and using your example only...

I have just added the `connectionTimeout: const Duration(seconds: 10)` in connect method as I seen you have suggested in [#354](https://github.com/PhilipsHue/flutter_reactive_ble/issues/354) and it's try to connect for 10 sec and then disconnect automatically... Attached the log for the same below.

```
D/BluetoothAdapter(28981): isLeEnabled(): ON
I/flutter (28981): Start connecting to 5C:B3:8D:1F:D9:F3
I/flutter (28981): ConnectionState for device 5C:B3:8D:1F:D9:F3 : DeviceConnectionState.connecting
D/BluetoothGatt(28981): connect() - device: 5C:B3:8D:1F:D9:F3, auto: false
D/BluetoothGatt(28981): registerApp()
D/BluetoothGatt(28981): registerApp() - UUID=3d28fecf-31fe-4848-997e-693613184cc6
D/BluetoothGatt(28981): onClientRegistered() - status=0 clientIf=8
I/flutter (28981): ConnectionState for device 5C:B3:8D:1F:D9:F3 : DeviceConnectionState.connecting
D/BluetoothManager(28981): getConnectionState()
D/BluetoothManager(28981): getConnectedDevices
D/BluetoothGatt(28981): close()
D/BluetoothGatt(28981): unregisterApp() - mClientIf=8
I/flutter (28981): ConnectionState for device 5C:B3:8D:1F:D9:F3 : DeviceConnectionState.disconnected 
```

Please let me know if you need any other information from my end.

remonh87, 2021-11-19:
Sorry for the late response I totally missed it. The logs show that the BluetoothGatt closes the connection  which indeed results in a `disconnectedState`. It looks like sth between the peripheral and device but I do not see any problem with the library.

closed 2021-11-19 by remonh87