PhilipsHue/flutter_reactive_ble #402: Backport to iOS 9

juliansteenbakker, 2021-10-05:
**Is your feature request related to a problem? (please describe)**
Right now the minimum iOS version is 11. For most people this isn't a problem however my implementation requires iOS 9 support. This is however achievable since the incompatibility right now is the usage of `CBManager` instead of `CBCentralManager`, and the functions of these two are the same. 

**Describe the solution you'd like**
You can use the following code to map `CBManager` to `CBCentralManager` :
```c
//#if __IPHONE_OS_VERSION_MAX_ALLOWED <= __IPHONE_9_3
#define CBManagerState CBCentralManagerState
#define CBManagerStateUnknown CBCentralManagerStateUnknown
#define CBManagerStateResetting CBCentralManagerStateResetting
#define CBManagerStateUnsupported CBCentralManagerStateUnsupported
#define CBManagerStateUnauthorized CBCentralManagerStateUnauthorized
#define CBManagerStatePoweredOff CBCentralManagerStatePoweredOff
#define CBManagerStatePoweredOn CBCentralManagerStatePoweredOn
//#endif
```

The only other function is a timer that needs to be converted.

```swift
    private func scheduleTaskTimeout(_ uniqueID: TaskQueue.Record.UniqueID, _ timeout: SubjectTask.Timeout) {
        if #available(iOS 10.0, *) {
            let timer = Timer.scheduledTimer(
                withTimeInterval: timeout.duration,
                repeats: false,
                block: papply(weak: self) { registry, _ in
                    registry.log("\(Controller.TaskSpec.tag) op timed out after \(timeout.duration) s, \(registry.tasks.count - 1) in queue")
                    timeout.handler()
                    registry.remove(uniqueID)
                }
            )
            scheduledTimeouts[uniqueID] = timer
        } else {
            //TODO: Implement 'old' timer function
            let timer = Timer.scheduledTimer(timeInterval: timeout.duration,
                                       target: self,
                                             selector: #selector(self.run(_:)),
                                     userInfo: nil,
                                      repeats: false)
            scheduledTimeouts[uniqueID] = timer
        }
        
    }
```

Once we've converted this, we can use this plugin on iOS 9 and higher. I tested it myself without the timer and i already got it working. The only thing i'm struggling with is how to convert the block from the `Timer.scheduledTimer` to a selector. Maybe someone has an idea?

Taym95, 2021-10-05:
If I understand correctly you want to backport the library to IOS with a workaround so it can support your implementation?

juliansteenbakker, 2021-10-06:
I understand, and of course it's up to you guys if you want this change. I think it depends on the use case of this package. Some developers focus on limiting supported iOS versions so they can provide better support to those. Others want to support a wider range of devices so they can maximize the audience for their application. The reason for me to support this is that clients that use my application still have old iPads which they can't just swap out that easy.

Again, if you don't want this change, just deny my PR and i'll keep the change in my local fork.

closed 2021-10-20 by remonh87