PhilipsHue/flutter_reactive_ble #408: Notifications/reads not working on iOS but they work in Android and in nRF Connect on iOS

JimTompkins, 2021-10-07:
**Describe the bug**
Notifications and reads not working on iOS but do work in nRF Connect and on Android

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one device using connectTo()
2. Subscribe to a characteristic (UUID 0xFFF4)
3. See messages being received in the console log but not being passed to the application

**Expected behavior**
Should work like on Android

With nRF Connect, when I turn on subscriptions I receive the notifications.

**Smartphone / tablet**
 - Device: iPhone 6
 - OS: iOS 12.5.4
 - Package version: master

**Peripheral device**
- Vendor, model: custom
- Does it run a custom firmware: yes

**Additional context**
Some comments about the log file below:
1) I can see that the characteristic with UUID = 0xFFF4 has kCBMsgArgCharacteristicProperties = 0x10 which I think indicates notify rather than read/write, etc.
2) I can see received messages with the expected argument data like this one:
debug	12:03:20.228857-0300	Runner	Received XPC message 95: {
    kCBMsgArgCharacteristicHandle = 39;
    kCBMsgArgData = <40>;
    kCBMsgArgDeviceUUID = "2D2F69D2-A2DA-4648-237B-59FD642DF052";
    kCBMsgArgTimestamp = 0;
}  but it doesn't make it through to the application.

** Console Log file**
default	12:03:05.582601-0300	Runner	flutter: HF: start BLE scan
debug	12:03:05.592781-0300	Runner	Sending XPC message 46: {
    kCBMsgArgOptions =     {
        kCBScanOptionAllowDuplicates = 1;
    };
    kCBMsgArgUUIDs =     (
        "0000FFF0-0000-1000-8000-00805F9B34FB"
    );
}
debug	12:03:06.132572-0300	Runner	Received XPC message 51: {
    kCBMsgArgAdvertisementData =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataServiceUUIDs =         (
            <fff0>
        );
    };
    kCBMsgArgDeviceUUID = "2D2F69D2-A2DA-4648-237B-59FD642DF052";
    kCBMsgArgName = HappyFeet;
    kCBMsgArgRssi = "-47";
}
debug	12:03:06.132894-0300	Runner	Received XPC message 51: {
    kCBMsgArgAdvertisementData =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = HappyFeet;
        kCBAdvDataServiceUUIDs =         (
            <fff0>
        );
        kCBAdvDataTxPowerLevel = "-3";
    };
    kCBMsgArgDeviceUUID = "2D2F69D2-A2DA-4648-237B-59FD642DF052";
    kCBMsgArgName = HappyFeet;
    kCBMsgArgRssi = "-47";
}
default	12:03:06.134342-0300	Runner	flutter: HF: found HappyFeet, connecting to device...
default	12:03:06.134520-0300	Runner	flutter: HF:    DiscoveredDevice(id: 2D2F69D2-A2DA-4648-237B-59FD642DF052, name: HappyFeet, serviceData: {}, serviceUuids: [fff0], manufacturerData: [], rssi: -47)
[... 273 lines cut ...]
    kCBMsgArgCharacteristicValueHandle = 46;
    kCBMsgArgData = <00>;
    kCBMsgArgDeviceUUID = "2D2F69D2-A2DA-4648-237B-59FD642DF052";
    kCBMsgArgType = 0;
}
debug	12:03:21.768323-0300	Runner	Received XPC message 98: {
    kCBMsgArgCharacteristicHandle = 38;
    kCBMsgArgDeviceUUID = "2D2F69D2-A2DA-4648-237B-59FD642DF052";
    kCBMsgArgResult = 0;
    kCBMsgArgState = 0;
}
debug	12:03:21.787887-0300	Runner	Received XPC message 96: {
    kCBMsgArgCharacteristicHandle = 45;
    kCBMsgArgDeviceUUID = "2D2F69D2-A2DA-4648-237B-59FD642DF052";
    kCBMsgArgResult = 0;
}
default	12:03:27.132714-0300	Runner	flutter: HF: disconnecting from device
debug	12:03:27.140727-0300	Runner	Sending XPC message 49: {
    kCBMsgArgDeviceUUID = "2D2F69D2-A2DA-4648-237B-59FD642DF052";
    kCBMsgArgState = 0;
}
debug	12:03:27.141660-0300	Runner	Received XPC message 53: {
    kCBMsgArgDeviceUUID = "2D2F69D2-A2DA-4648-237B-59FD642DF052";
    kCBMsgArgResult = 0;
}

JimTompkins, 2021-10-13:
I tried to use reactive_ble_mobile 4.0.1 but the version solving fails.  So I removed the dependency override on reactive_ble_moble.

After doing a conversion to Swift 5, I get this reduced list of errors:
[image]
:

Should 5.0.0 work with Xcode 12.4?

Thanks for any hints.

remonh87, 2021-10-16:
@JimTompkins version 4.0.1 and higher only support xcode13. Reason is that Apple introduced some breaking changes in `CBCharacteristic` (the value `service` is now nullable).  In order to make REactiveble work is to upgrade to XCode13 or use the dependency override. This override does not work for version 5.0.0 since we introduced some breaking changes (which is why we introduced a major version bump). I will update the readme so it is a bit more clear