PhilipsHue/flutter_reactive_ble #417: how to resolve connect multi device crashed

liang-notes, 2021-10-19:
**Describe the bug**
1. using connectToDevice to connect a device
2. then using connectToDevice to connect another device
3. crashed

reactive_ble_mobile/ConnectTaskController.swift:16: Assertion failed
* thread #1, queue = 'com.apple.main-thread', stop reason = Assertion failed
    frame #0: 0x000000019aac0060 libswiftCore.dylib`_swift_runtime_on_report
libswiftCore.dylib`_swift_runtime_on_report:
->  0x19aac0060 <+0>: ret
libswiftCore.dylib`_swift_reportToDebugger:
    0x19aac0064 <+0>: b      0x19aac0060               ; _swift_runtime_on_report
libswiftCore.dylib`_swift_shouldReportFatalErrorsToDebugger:
    0x19aac0068 <+0>: adrp   x8, 324926
    0x19aac006c <+4>: ldrb   w0, [x8, #0xfd1]

**To Reproduce**
Steps to reproduce the behavior:
1. using connectToDevice to connect a device
2. then using connectToDevice to connect another device
3. crashed

**Expected behavior**
A clear and concise description of what you expected to happen.

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: iPhone 8P
 - OS: iOS 13,
 - Package version: 5.0.0

**Peripheral device**
- Vendor, model: [e.g. Fitbit Versa, iPhone SE, or CUSTOM]
- Does it run a custom firmware / software: yes / no

**Additional context**
Add any other context about the problem here.

remonh87, 2021-10-31:
@aBuder  that is correct you need to update to xcode13

remonh87, 2021-11-08:
I think this is related to #439 there is a pr on the way

remonh87, 2021-11-08:
fixed with version 5.0.2

closed 2021-11-08 by remonh87