PhilipsHue/flutter_reactive_ble #431: Scan Failure Location Permission missing

LeChatNoire93, 2021-10-30:
**Describe the bug**

No connection possible with the other Device.

Console says:
I/flutter (17422): Exception: GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Location Permission missing (code 3)")

Despite having added:
    <uses-permission-sdk-23
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        android:maxSdkVersion="30" />
    <uses-permission-sdk-23
        android:name="android.permission.ACCESS_FINE_LOCATION"
        android:maxSdkVersion="30" />

in the Android Manifest.

**To Reproduce**
Steps to reproduce the behavior:
updated to the newes version of flutter_reactive_ble

**Expected behavior**
connection to arduino ble and reading some services

**Smartphone / tablet**
 - Device: Xiaomi Redmi Note 10 Pro
 - OS: Android 11
 - MIUI 12.5.4

**Peripheral device**
- Vendor, model: Arduino Nano BLE Sense
- Does it run a custom firmware / software:  no
the arduino sketch works, I have an python script which is still able to connect.
By the way the flutter app worked before fine.

remonh87, 2021-10-31:
if you set the `requireLocationServicesEnabled` to false you will not get this error. Check the documentation of this function for the consequences and determine if you need this parameter set to either true or false. 

If you need to set it to true make sure the location permission is granted.

closed 2021-10-31 by remonh87

GaelleJoubert, 2021-11-03:
I have a similar error. `requireLocationServicesEnabled` is set to false. 
I have to add `<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />`  in the android manifest, and explicitely ask for the permission in my code.

My issue is that when I try to uplaud my code in android play console I have an error saying there is a two declaration of  ACCESS_FINE_LOCATION with different max SDK. It looks like the authorisation in manifest and the package manifest are conflicting. 
How can I remove the one from the package ? 
(note: I'm building for SDK 31)
Thanks !

orestesgaolin, 2021-11-08:
Similarly, if I don't set it in my own AndroidManifest, the value from the plugin is not merged with the app's manifest.