PhilipsHue/flutter_reactive_ble #434: Android 12 - Bluetooth cannot start (code 0)

GaelleJoubert, 2021-11-02:
I had a code doing a simple bluetooth scan and it  worked, until I upgraded my phone with Android 12.

Here is the code where i call the scan : 

```
flutterReactiveBle.scanForDevices(withServices: [mask], scanMode: ScanMode.balanced, requireLocationServicesEnabled: false).listen((device) async {
//some irrelevent data treatement here}, 
onError: (e)=> handleScanFailure(e),
  );
```

The log of the error I'm having is the following : 

_I/flutter ( 3345): Ble2 : BleStatus.ready
D/BluetoothAdapter( 3345): isLeEnabled(): ON
D/BluetoothAdapter( 3345): isLeEnabled(): ON
D/BluetoothLeScanner( 3345): could not find callback wrapper
I/flutter ( 3345): Exception: GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Bluetooth cannot start (code 0)")_

I do have all the permission in my manifest :    
```
<uses-permission android:name="android.permission.BLUETOOTH" />
 <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
 <uses-permission android:name="android.permission.BLUETOOTH_SCAN"
        android:usesPermissionFlags="neverForLocation" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
        android:maxSdkVersion="30" />

    <uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
```

My buildgraddle use the following SDK version : 

```
compileSdkVersion 31
minSdkVersion 29 
targetSdkVersion 30
```

So **the bluetooth is enable and authorised**, but the scan doesn't wants to launch .. any idea what could be wrong ?

**Smartphone / tablet**
 - Device: Google Pixel 4A 5G, Android 12

GaelleJoubert, 2021-11-03:
EDIT : 
I found out that I had to manually ask for BLUETOOTH_SCAN AND BLUETOOTH_CONNECT.
AND that I need the <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />.

My issue now is that in the this package manifest ACCESS_FINE_LOCATION is set with max SDK 30. But I do need it for SDK 31 as well. 
When I try to upload my app bumdle in android console for a internal release, I have the error : 
`"Duplicate declarations of permission android.permission.ACCESS_FINE_LOCATION with different maxSdkVersions."`

I guess that is the manifest of the package merged with mine. How can I remove the `maxSdkVersion = 30 `from the package manifest ?

PieterAelse, 2021-11-03:
@GaelleJoubert You can use `tools:remove="android:maxSdkVersion"` inside the permission for that. E.G:
```xml
<uses-permission-sdk-23
    android:name="android.permission.ACCESS_FINE_LOCATION"
    tools:remove="android:maxSdkVersion" />
```

(you need to add the namespace too if you didn't use `tools` before:
```xml
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:tools="http://schemas.android.com/tools"
    ...
```

GaelleJoubert, 2021-11-03:
Thanks for the fast answer. 
I tried that in my app manifest, but I still have the same error from the android console : 
`
Duplicate declarations of permission android.permission.ACCESS_FINE_LOCATION with different maxSdkVersions.`

This is my manifest (up to the important part) : 

```
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="fr.kip.app"
    xmlns:tools="http://schemas.android.com/tools">

    <queries>
        <!-- If your app makes calls -->
        <intent>
            <action android:name="android.intent.action.DIAL" />
            <data android:scheme="tel" />
        </intent>
    </queries>

    <uses-permission android:name="android.permission.INTERNET"/>
    <!-- required for API 18 - 30 -->
    <uses-permission
        android:name="android.permission.BLUETOOTH"
        android:maxSdkVersion="30" />
    <uses-permission
        android:name="android.permission.BLUETOOTH_ADMIN"
        android:maxSdkVersion="30" />

    <!-- API 31+ -->
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>

    <uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"
        android:maxSdkVersion="30"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"
        tools:remove="android:maxSdkVersion"/>
```

Do you see any mistakes ?

PieterAelse, 2021-11-03:
@GaelleJoubert maybe apply the same tools:remove to the COARSE_LOCATION?

One other difference I spot with e.g. our configuration is `android:usesPermissionFlags="neverForLocation"` for `BLUETOOTH_SCAN`:
```xml
<uses-permission
    android:name="android.permission.BLUETOOTH_SCAN"
    android:usesPermissionFlags="neverForLocation"
    tools:targetApi="s" />
```

Maybe that makes a difference in manifest merging 🤔 

For example, this is a bigger manifest snipped we have, which compiles without issues:
<details>
<summary>AndroidManifest.xml</summary>
<blockquote>

     
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission
        android:name="android.permission.BLUETOOTH_SCAN"
        android:usesPermissionFlags="neverForLocation"
        tools:targetApi="s" />
    

    

    
    <uses-permission-sdk-23
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        tools:remove="android:maxSdkVersion" /> 
    <uses-permission-sdk-23
        android:name="android.permission.ACCESS_FINE_LOCATION"
        tools:remove="android:maxSdkVersion" /> 

    
</blockquote>
</details>

GaelleJoubert, 2021-11-03:
Thanks !
I added that flag, and also changed `<uses-permission` into `<uses-permission-sdk-23` for the locations permissions like you did in your exemple, and it works now ! :D

PieterAelse, 2021-11-03:
Good to hear! So I can close this ticket? 😄

GaelleJoubert, 2021-11-03:
Yes, thanks !

closed 2021-11-03 by GaelleJoubert