PhilipsHue/flutter_reactive_ble #435: [Question] What will happen to this plugin now that Polidea stoped developing code?

Henge9, 2021-11-02:
Hi.
The flutter plugin flutter_ble_lib from Polidea stopped getting upgrades after Polidea got bought up and stopped developing code. So I am looking for a new BLE package and got this one recommended by Google.

The thing that worries me is that this plugin uses nativ plugins from Polidea. Example: com.polidea.rxandroidble2. 

Are Polideas nativ plugins still alive? Will this plugin survive anyway?

remonh87, 2021-11-02:
This library is the heart of the Hue Bluetooth app so yes we are committed to keep this plugin alive. We only  use a Polidea library for Android. Reason is that they did an awesome job working around the numerous issues on BLE for Android. I noticed the maintenance time on their side dropped but if necessary we would fork the existing library. At the moment there is one maintainer on the Android side that is doing as much as he can to keep up to date (e.g. he introduced the new Android 12 permissions).

closed 2021-11-02 by remonh87

Henge9, 2021-11-02:
Fantastic. Thank you for the fast answer. That feels good to hear. Now I can convince my boss to give me time to change BLE plugin again, for the last time.