PhilipsHue/flutter_reactive_ble #441: Required Kotlin version is higher than the default version used by Kotlin

orestesgaolin, 2021-11-08:
**Describe the bug**

When building the Flutter app with the plugin with stable Flutter following error is seen:

> e: Incompatible classes were found in dependencies. Remove them from the classpath or use '-Xskip-metadata-version-check' to suppress errors
> e: /Users/dominik/.gradle/caches/transforms-2/files-2.1/241aa167d59e4b859ad3918715ac2f3b/jetified-kotlin-stdlib-1.5.31.jar!/META-INF/kotlin-stdlib.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.5.1, expected version is 1.1.15.

The required version 1.5.31 is visible [here in build.gradle](https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/packages/reactive_ble_mobile/android/build.gradle#L6).

**Expected behavior**

Either the version should be the same as the default Kotlin version used by Flutter project (1.3.50) or please note the required change in `build.gradle` in the README.md.

**Smartphone / tablet**
 - Device: Android
 - OS: Android 12
 - Package version: 5.0.1
 
```
[✓] Flutter (Channel stable, 2.5.3, on macOS 12.0.1 21A559 darwin-x64, locale pl-PL)
    • Flutter version 2.5.3 at /Users/dominik/fvm/versions/stable
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 18116933e7 (3 weeks ago), 2021-10-15 10:46:35 -0700
    • Engine revision d3ea636dc5
    • Dart version 2.14.4

[✓] Android toolchain - develop for Android devices (Android SDK version 31.0.0)
    • Android SDK at /Users/dominik/Library/Android/sdk
    • Platform android-31, build-tools 31.0.0
    • ANDROID_HOME = /Users/dominik/Library/Android/sdk
    • Java binary at: /Applications/Android Studio.app/Contents/jre/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7281165)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 13.1, Build version 13A1030d
    • CocoaPods version 1.11.2

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 2020.3)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7281165)

[✓] VS Code (version 1.62.0)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.28.0

[✓] Connected device (3 available)
    • SM G991B (mobile) •xx • android-arm64  • Android 12 (API 31)
    • macOS (desktop)   • macos       • darwin-x64     • macOS 12.0.1 21A559 darwin-x64
    • Chrome (web)      • chrome      • web-javascript • Google Chrome 95.0.4638.69

• No issues found!
```

pacifi5t, 2021-11-08:
I have faced the same issue. To fix this quickly change Kotlin version to 1.4.32 for now

Taym95, 2021-11-08:
Thanks @orestesgaolin, we will add that to readme, for now you can just updated kotlin

remonh87, 2021-11-11:
hmm actually this is a good point I see no reason for us to bump the Kotlin version, besides only to get rid of some warnings in Android studio. Let's use the Flutter default

Taym95, 2021-11-15:
@remonh87  I see that the flutter team [bumped](https://github.com/flutter/flutter/pull/92052) kotlin version to `'1.5.31'` , I see no reason to downgrade it ?

remonh87, 2021-11-15:
no let's add it to the readme for now indeed

closed 2021-12-01 by Taym95