PhilipsHue/flutter_reactive_ble #442: Not able to run on Android with version 5.0.1

albo1337, 2021-11-09:
I am not able to run the app on Android with version 5.0.1. The Problem seems to be with the new testing libraries:

* What went wrong:
A problem occurred configuring project ':reactive_ble_mobile'.
> class com.android.build.gradle.internal.api.TestVariantImpl_Decorated cannot be cast to class com.android.build.gradle.internal.core.InternalBaseVariant (com.android.build.gradle.internal.api.TestVariantImpl_Decorated is in unnamed module of loader org.gradle.internal.classloader.VisitableURLClassLoader @10cf6863; com.android.build.gradle.internal.core.InternalBaseVariant is in unnamed module of loader org.gradle.internal.classloader.VisitableURLClassLoader @54e0cba7)

Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 2.5.3, on macOS 11.5.2 20G95 darwin-x64, locale de-CH)
[✓] Android toolchain - develop for Android devices (Android SDK version 30.0.3)
[✓] Xcode - develop for iOS and macOS
[✓] Chrome - develop for the web
[✓] Android Studio (version 2020.3)
[✓] VS Code (version 1.57.1)
[✓] Connected device (3 available)

• No issues found!

Does anyone else know what to do?

yuriboeira11tx, 2021-11-09:
Try to replicate the settings present in the library example application, some information has changed.

albo1337, 2021-11-10:
Nice. I had to update gradle properties, and set compile and target sdk do newest version. What I also had to do was adding android:exported="true" in my AndroidManifest.xml for all activities and receiver.

Thank you @yuriboeira11tx

closed 2021-11-10 by albo1337