PhilipsHue/flutter_reactive_ble #443: Can not read charecteristic

albo1337, 2021-11-09:
I am not able to read characteristic 

discovered device DiscoveredDevice(id: 5A449B54-B133-A6D5-2DCB-2F47C4F68B66, name: MY-BUTTON 254m, serviceData: {}, serviceUuids: [2168cb00-e9e4-475b-a14b-1ed34d0c05ce], manufacturerData: [], rssi: -37)

connected to device ConnectionStateUpdate(deviceId: 5A449B54-B133-A6D5-2DCB-2F47C4F68B66, connectionState: DeviceConnectionState.connected, failure: null)

Warning! No subscription to report a characteristic read failure: A peripheral 5A449B54-B133-A6D5-2DCB-2F47C4F68B66 is unknown (make sure it has been discovered)

Why this happens? What do I have to do to ensure, that I can read characteristics?

albo1337, 2021-11-10:
When I use `subscribeToCharacteristic` I am facing an error on Android, maybe this could help to find the problem:

error is Exception: GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "failed to connect Already connected to device with MAC address 68:0A:E2:F1:88:B1")

Do you guys maybe now what I am doing wrong?

Taym95, 2021-11-15:
Can you please follow [the issue template](https://github.com/PhilipsHue/flutter_reactive_ble/issues/new?assignees=&labels=&template=bug-report.md&title=) so we can help, also a code example will be good!?

albo1337, 2021-11-15:
I guess I got my problem fixed yesterday evening. Thank you @Taym95 for trying to help out!

closed 2021-11-15 by albo1337

simranss, 2023-05-09:
@albo1337 How did you fix it?