PhilipsHue/flutter_reactive_ble #445: Can no longer build for iOS

wabash9000, 2021-11-11:
Everything built for Android without issue, pushed to my GitLab and pulled on my MacBook. Run "flutter run --release" to install on my test phone and get a bunch of errors in terminal all related to this package.  I've tried my best to update everything to the newest libraries, run "flutter clean", "flutter pub get", "flutter pub update", and "flutter pub outdated" to try to clean up and track down any issues.

Errors I get are:
/Library/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:80:44: error: cannot force unwrap value of non-optional type 'CBPeripheral'
                        key: service.peripheral!.identifier,
    /Library/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:230:54: error: cannot use optional chaining on non-optional value of type 'CBService'
            guard let peripheral = characteristic.service?.peripheral
    /Library/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:255:54: error: cannot use optional chaining on non-optional value of type 'CBService'
            guard let peripheral = characteristic.service?.peripheral
    /Library/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:273:52: error: cannot use optional chaining on non-optional value of type 'CBService'
            guard let response = characteristic.service?.peripheral?.writeValue(value, for: characteristic, type: .withoutResponse)
    /Library/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:273:64: error: cannot use optional chaining on non-optional value of type 'CBPeripheral'
            guard let response = characteristic.service?.peripheral?.writeValue(value, for: characteristic, type: .withoutResponse)
    /Library/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Tasks/CharacteristicNotify/CharacteristicNotifyTaskController.swift:14:54: error: cannot use optional
    chaining on non-optional value of type 'CBService'
            guard let peripheral = characteristic.service?.peripheral
    /Library/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/Plugin/PluginController.swift:293:50: error: cannot use optional chaining on non-optional value of type
    'CBService'
                            if characteristic.service?.uuid.data != nil {
    /Library/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/Plugin/PluginController.swift:294:86: error: cannot force unwrap value of non-optional type 'CBService'
                                $0.serviceID = Uuid.with{$0.data = characteristic.service!.uuid.data}
    /Library/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/BleData extras/QualifiedCharacteristic.swift:16:46: error: cannot use optional chaining on non-optional value of
    type 'CBService'
                serviceID: characteristic.service?.uuid ?? ServiceID(),
    /Library/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/BleData extras/QualifiedCharacteristic.swift:17:49: error: cannot use optional chaining on non-optional value of
    type 'CBService'
                peripheralID: characteristic.service?.peripheral?.identifier ?? PeripheralID(uuid: (0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0))
    /Library/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/BleData extras/QualifiedCharacteristic.swift:17:61: error: cannot use optional chaining on non-optional value of
    type 'CBPeripheral'
                peripheralID: characteristic.service?.peripheral?.identifier ?? PeripheralID(uuid: (0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0))

Output from "flutter doctor"
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 2.5.0, on macOS 11.5.2 20G95 darwin-x64, locale en-US)
[✗] Android toolchain - develop for Android devices
    ✗ Unable to locate Android SDK.
[✓] Xcode - develop for iOS and macOS
[✓] Chrome - develop for the web
[✓] Android Studio (version 2020.3)
[✓] VS Code (version 1.61.0)
[✓] Connected device (2 available)
Don't build for android on this device so didn't think I would need the android SDK.  All of the errors seem to be in the Swift portion of the library.

wabash9000, 2021-11-11:
Not sure if it is related but my Xcode is out of date. Working on the painful process of updating it right now.

remonh87, 2021-11-11:
The latest version needs Xcode 13. Apple introduced some breaking changes so that's why we are forced to upgrade it

closed 2021-11-11 by remonh87