PhilipsHue/flutter_reactive_ble #446: BLE State is unknown when code runs in the background

ahoelzemann, 2021-11-11:
Hey, 

I'm trying to send data from a smart watch to the phone using your library. This task takes up to 45 minutes, depending on the amount of data stored on the smart watch. Since this task runs for such a long time, it is necessary to let it run in the background.  I'm using the plugin flutter_isolate for this. 

Whenever I'm running my code in an isolate (on an iPhone X), the BLE state is set to unknown. However, I would like to check before connecting to the device that BLE is ready. The rest of the code works so far and I can actually ignore it, however it would be cleaner to check the state. Any idea why this is happening? 

Method with which I'm checking the BLE state.

`  Future<dynamic> _checkIfBLAndLocationIsActive() {
    Completer completer = new Completer();
    if (Platform.isAndroid) {
      if (_ble.status == BleStatus.locationServicesDisabled) {
        completer.complete("locationPoweredOff");
      }
    }
    if (_ble.status == BleStatus.poweredOff) {
      completer.complete("blePoweredOff");
    }
    if (_ble.status == BleStatus.ready) {
      completer.complete("ready");
    }
    if (_ble.status == BleStatus.unknown) {
      completer.complete("ready");
    }
    return completer.future;
  }`

freedreamer82, 2021-11-17:
same here.
I run the lib in an isolate and it seems it does not running well , i cannot have events from BLE status as well.

M.

ahoelzemann, 2021-11-18:
The problem was my method to check if BLE is ready. I din't follow the example. I checked the status variable on my instance of FlutterReativeBle , however you really need to use the listener on the statusStream. 

I'm not sure if the delay is really necessary, but I wanted to make sure that the underlaying API had time to be set to ble.ready.

```
 Future<dynamic> _checkIfBLAndLocationIsActive() async {
    Completer completer = new Completer();
    _ble.statusStream.listen((event) async {
      if (event == BleStatus.ready) {
        await Future.delayed(Duration(milliseconds: 1500));
        completer.complete("ready");
    });
    return completer.future;
  }
```

closed 2021-11-18 by ahoelzemann