PhilipsHue/flutter_reactive_ble #447: Unable to run example on iOS simulator

omartinma, 2021-11-12:
**Describe the bug**
Just cloned the repo and run your example and I get iOS build errors.

**To Reproduce**
Steps to reproduce the behavior:

1. Clone example.
2. Run on iOS simulator.
3. See the logs

<details>
  <summary>Logs</summary>

```
Launching lib/main.dart on iPhone 12 Pro Max in debug mode...
lib/main.dart:1
Xcode build done.                                           21,8s
Failed to build iOS app
Error output from Xcode build:
↳
    ** BUILD FAILED **
Xcode's output:
↳
    /Users/oscarmartin/Repos/flutter/flutter_reactive_ble/packages/reactive_ble_mobile/ios/Classes/ReactiveBle/Central.swift:80:44: error: cannot force unwrap value of non-optional type 'CBPeripheral'
                        key: service.peripheral!.identifier,
                             ~~~~~~~~~~~~~~~~~~^
    /Users/oscarmartin/Repos/flutter/flutter_reactive_ble/packages/reactive_ble_mobile/ios/Classes/ReactiveBle/Central.swift:230:54: error: cannot use optional chaining on non-optional value of type 'CBService'
            guard let peripheral = characteristic.service?.peripheral
                                   ~~~~~~~~~~~~~~~~~~~~~~^
    /Users/oscarmartin/Repos/flutter/flutter_reactive_ble/packages/reactive_ble_mobile/ios/Classes/ReactiveBle/Central.swift:255:54: error: cannot use optional chaining on non-optional value of type 'CBService'
            guard let peripheral = characteristic.service?.peripheral
                                   ~~~~~~~~~~~~~~~~~~~~~~^
    /Users/oscarmartin/Repos/flutter/flutter_reactive_ble/packages/reactive_ble_mobile/ios/Classes/ReactiveBle/Central.swift:258:37: error: type of expression is ambiguous without more context
            characteristicWriteRegistry.updateTask(
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
    /Users/oscarmartin/Repos/flutter/flutter_reactive_ble/packages/reactive_ble_mobile/ios/Classes/ReactiveBle/Central.swift:273:52: error: cannot use optional chaining on non-optional value of type 'CBService'
            guard let response = characteristic.service?.peripheral?.writeValue(value, for: characteristic, type: .withoutResponse)
                                 ~~~~~~~~~~~~~~~~~~~~~~^
    /Users/oscarmartin/Repos/flutter/flutter_reactive_ble/packages/reactive_ble_mobile/ios/Classes/ReactiveBle/Central.swift:273:64: error: cannot use optional chaining on non-optional value of type 'CBPeripheral'
            guard let response = characteristic.service?.peripheral?.writeValue(value, for: characteristic, type: .withoutResponse)

```

```
➜  flutter_reactive_ble git:(master) ✗ flutter doctor -v
[✓] Flutter (Channel stable, 2.5.3, on Mac OS X 10.15.7 19H1323 darwin-x64, locale es-ES)
    • Flutter version 2.5.3 at /Users/oscarmartin/SDKS/flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 18116933e7 (4 weeks ago), 2021-10-15 10:46:35 -0700
    • Engine revision d3ea636dc5
    • Dart version 2.14.4

[!] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
    • Android SDK at /Users/oscarmartin/Library/Android/sdk
    ✗ cmdline-tools component is missing
      Run `path/to/sdkmanager --install "cmdline-tools;latest"`
      See https://developer.android.com/studio/command-line for more details.
    ✗ Android license status unknown.
      Run `flutter doctor --android-licenses` to accept the SDK licenses.
      See https://flutter.dev/docs/get-started/install/macos#android-setup for more details.

[✓] Xcode - develop for iOS and macOS
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 12.1, Build version 12A7403
    • CocoaPods version 1.11.2

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 3.4)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin version 39.0.1
    • Dart plugin version 183.6270
    • Java version OpenJDK Runtime Environment (build 1.8.0_152-release-1343-b01)

[✓] VS Code (version 1.62.1)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.28.0

[✓] Connected device (2 available)
    • iPhone 12 Pro Max (mobile) • 978A339D-9CD7-4916-81D7-7C60A4CE65CC • ios            •
      com.apple.CoreSimulator.SimRuntime.iOS-14-1 (simulator)
    • Chrome (web)               • chrome                               • web-javascript • Google Chrome 95.0.4638.69

! Doctor found issues in 1 category.
➜  flutter_reactive_ble git:(master) ✗ 
```

</details>

remonh87, 2021-11-12:
duplicate of https://github.com/PhilipsHue/flutter_reactive_ble/issues/445

closed 2021-11-12 by remonh87

iameerhamza, 2021-11-22:
> 

Hello, i am facing the same issue, can you help me that how did you resolve that issue.
Thanks in Advance..