PhilipsHue/flutter_reactive_ble #449: blocked in readCharacteristic when connection is lost (iOS only)

fabione2000, 2021-11-15:
**Describe the bug**
When disconnection occurs (from remote device) during readCharacteristic() call, readCharacteristic doesn't return and doesn't throw exception.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one device using `connectToDevice()'
2. Discover services using discoverServices
3. Read characteristic using readCharacteristic()
4. Power off the connected device.
5. the ConnectionStateUpdate stream notify new status 'disconnected'
--> readCharacteristic() doesn't return an empty List<int> as expected. No exception neither.

My readCharacteristic function (idem sample)

  Future<List<int>> readCharacteristic(
      QualifiedCharacteristic characteristic) async {
    try {
      final result = await _readCharacteristic(characteristic);

      print('Read ${characteristic.characteristicId}: value = $result');
      return result;
    } on Exception catch (e, s) {
      print(
        'Error occured when reading ${characteristic.characteristicId} : $e',
      );
      // ignore: avoid_print
      print(s);
      rethrow;
    }
  }

**Expected behavior**
readCharacteristic() returns when the disconnection occurs.
This behaviour is on Android.

**Smartphone / tablet**
 - Device:  iPhone 12
 - OS: iOS 15.0.2 
 - Ble Package version: 5.0.2

**Peripheral device**
- Vendor, model: CUSTOM
- Does it run a custom firmware / software: yes

**Additional context**
Add any other context about the problem here.

yuriboeira11tx, 2021-11-18:
Along with that, are you having problem with ```subscribeToCharacteristic()``` on iOS too?

remonh87, 2021-11-19:
@fabione2000  this is difficult to reproduce since reads are generally quite fast. That said I would implement some custom logic in your app that if an operation is running and you receive a disconnect to throw an Exception. I think returning an empty list is confusing. I am surprised Android is working like this. I would have expected it to be running forever as well.

closed 2021-12-15 by remonh87