PhilipsHue/flutter_reactive_ble #450: Request-Response approach

albo1337, 2021-11-16:
**Is your feature request related to a problem? (please describe)**
I have a problem, while trying to achieve a request-response mechanism.

**Describe the solution you'd like**
Request-Response mechanism that works properly

**Additional context**
I have already tried to achieve a request-response mechanism. This ist what I have until now:

```
Future sendRequest(QueryCommand command, Query query, {int currentTry = 1, int maxTry = 2}) {
    var response = Rx.combineLatest2<List<int>, String, QueryCommandResult>(
      _responseMonitor(),
      _write(command),
      (response, requestResult) {
        final result = String.fromCharCodes(Uint8List.fromList(response)).substring(command.bleCommandASCII.length);
        QueryCommandResult queryCommandResult = QueryCommandResult(result, command, query);
        return queryCommandResult;
      },
    )
        .timeout(Duration(seconds: 3), onTimeout: (eventSink) {
          eventSink.close();
          return null;
        })
        .first
        .catchError((error) {
          Fimber.e("Has error: $error");
          return null;
        });

    return Future.value(response);  
  }

  Stream _write(QueryCommand command) {
    QualifiedCharacteristic qualifiedCharacteristic = QualifiedCharacteristic(
      characteristicId: _requestCharacteristic.characteristicId,
      serviceId: _requestCharacteristic.serviceId,
      deviceId: discoveredDevice.id,
    );
    StreamController<String> controller;
    controller = StreamController<String>(onListen: () async {
      await _flutterReactiveBle
          .writeCharacteristicWithResponse(
        qualifiedCharacteristic,
        value: command.commandToWrite().map((e) => int.parse(e)).toList(),
      )
          .catchError((error) {
        Fimber.e("Error $error");
      });
      controller.add(command.bleCommandASCII);
      controller.close();
    });
    return controller.stream;
  }

  Stream<List<int>> _responseMonitor() {
    QualifiedCharacteristic qualifiedCharacteristic = QualifiedCharacteristic(
      characteristicId: _responseCharacteristic.characteristicId,
      serviceId: _responseCharacteristic.serviceId,
      deviceId: discoveredDevice.id,
    );
    return _flutterReactiveBle.subscribeToCharacteristic(qualifiedCharacteristic);
  }
}
```

The problem is, that it only works once and if I start a request for another query it won't work anymore. Does anyone maybe see the problem in my code here?

albo1337, 2021-11-16:
If I change the method _responseMonitor to this:

```
Stream<List<int>> _responseMonitor() {
    QualifiedCharacteristic qualifiedCharacteristic = QualifiedCharacteristic(
      characteristicId: _responseCharacteristic.characteristicId,
      serviceId: _responseCharacteristic.serviceId,
      deviceId: discoveredDevice.id,
    );
    StreamSubscription<List<int>> streamSubscription;
    Stream stream = _flutterReactiveBle.subscribeToCharacteristic(qualifiedCharacteristic);
    streamSubscription = stream.listen((event) {
      streamSubscription.cancel();
    });
    return stream;
  }
```

I get the following error: Unhandled Exception: Bad state: Stream has already been listened to.

I guess I have to handle the subscribeToCharacteristic different, but I am absolutly not knowing how

albo1337, 2021-11-16:
I've found a solution for this approach:

```
_responseMonitor() {
    QualifiedCharacteristic qualifiedCharacteristic = QualifiedCharacteristic(
      characteristicId: _responseCharacteristic.characteristicId,
      serviceId: _responseCharacteristic.serviceId,
      deviceId: discoveredDevice.id,
    );

    _flutterReactiveBle.subscribeToCharacteristic(qualifiedCharacteristic).listen((event) {
      mySteamInputSink.add(event);
    });
  }

  var _myController = BehaviorSubject<List<int>>();
  Sink<List<int>> get mySteamInputSink => _myController.sink;
  Stream<List<int>> get myStream => _myController.stream;
```

Every time I call the function "sendRequest" I have to do the following:

```
_myController.close();
_myController = BehaviorSubject<List<int>>();
```

_responseMonitor is getting called as soon as a connection with the device has been established. With this way I am able to acchieve a request-response mechanism. Is there someone who is seeing trouble in my code?

Taym95, 2021-11-16:
First of all this place for posting issues related to  `flutter_reactive_ble` library, would be great if one of the maintainers help you our but I would advise you to ask this type of question on [stackoverflow](https://stackoverflow.com/), I close this issue for now.

closed 2021-11-17 by albo1337