PhilipsHue/flutter_reactive_ble #455: Example Project for Getx Statemanagment Users , with Easiest Implementation

rohitsangwan01, 2021-11-18:
**Describe the solution you'd like**
Hey , its a Great package , but few Users still getting issue in Running or Working with  this Package , so i added a Sample ready to use Project by using Getx Reactive Statemanagement , and also planning to record a Youtube Video , for helping others

https://github.com/rohitsangwan01/flutter_bluetooth

its my repo , and if you guys found it usefull , you can add it in unofficial Demo project , Thankyou

remonh87, 2021-11-18:
Thanks it is always good to show some example projects I will check it out.

Taym95, 2021-11-20:
I checked the app, looks great thanks, but contains only scan and connect and mainly focuses on `Getx` implementation not much about the `flutter_reactive_ble` library

rohitsangwan01, 2021-11-21:
@Taym95 , @remonh87 ,Thanks , yeah that was just a sample of code from my Main App building for Nerbio, thats fully based on this library ,that have almost everything with reactive UI approach by Getx, so can u tell me how i can improve this Example or what else i can add ,to be more helpfull for other Peoples, i wanted to create a youtube video tutorial about this package + Getx + NordicDfu ( https://pub.dev/packages/nordic_dfu ) for pushing Dfu updates to device ,

remonh87, 2021-11-23:
@rohitsangwan01  I would use a very minimalistic example using getx containing: scanning, connecting, checking ble status, read /write /subscribe to a characteristic.  I think the focus should be on using BLE and not the fancy GETX components.

About your question regarding DFU. We won't add this since this is very chip specific. IF we start on this road we will get questions from people using other vendors. I think it is better to use a community example and refer this in our readme so people can checkout how to use DFU for Nordic chips.

remonh87, 2021-12-04:
I close this issue now feel free to ping me when the example is done.

closed 2021-12-04 by remonh87