PhilipsHue/flutter_reactive_ble #459: Is there any option to get characteristic data in form of UTF-8 Or As Char on behalf of List<Int>?

DhavalRKansara, 2021-11-22:
I am using the example available under this repository for to establish connection and read the data from my bluetooth device...

The problem with this package is it provide data in form of List<int> as you can show here -  [` late StreamSubscription<List<int>>? subscribeStream;`](https://github.com/PhilipsHue/flutter_reactive_ble/blob/a78dc4ea6201ad80f0f020783d22cd3f19af9e6b/example/lib/src/ui/device_detail/characteristic_interaction_dialog.dart#L62)

I have tried to perform the same operation to read the data by establishing the subscription using the [Bluetooth Le Scanner](https://play.google.com/store/apps/details?id=uk.co.alt236.btlescan&hl=en_IN&gl=US) app and it provide the data in form of **as array + as UTF - 8 + as char**... while using this package I can get the data in form List<int> only... You can check the below screenshot of[ Bluetooth Le Scanner](https://play.google.com/store/apps/details?id=uk.co.alt236.btlescan&hl=en_IN&gl=US) for reference....

[image]

Please let me know if I miss any solution which is already present for my issue and I missed it and also let me know if you need any further information from my end. 

Thanks in Advance.

Taym95, 2021-11-22:
Hi, Converting to `UTF - 8` or  `Char `  is logic that needs to be implemented inside your app, `Bluetooth LE Scanner` is an app that adds this logic and this is a library, we don't want to force the conversing inside the library, eg: you can do this inside your app:

```
List<int> charCodes = [97, 98, 99, 100];
 String result = utf8.decode(charCodes); 
```

werediver, 2021-11-23:
@Taym95 This is not a library issue, indeed. Thanks for suggesting a solution to the issue author.

@DhavalRKansara Taym's example makes use of [utf8 top-level constant](https://api.flutter.dev/flutter/dart-convert/utf8-constant.html) from `dart:convert` library (a part of the Dart SDK).

closed 2021-11-23 by werediver