PhilipsHue/flutter_reactive_ble #462: Stop scan for devices

Nelmajuva, 2021-11-28:
Hi, thank you very much for taking the time to read this.

I am using this package to connect to a watch via bluetooth and so far everything is going great.

Once I scan, I already have my device, but how once the device is detected can I stop searching for other devices?

`
_flutterReactiveBle.scanForDevices(
  withServices: [_deviceUuid],
  requireLocationServicesEnabled:  false)
    .listen((DiscoveredDevice device) {
      print(device);
  });
`

Nelmajuva, 2021-11-28:
Additional: It is evident that it always returns the same device, but it does it indefinitely n number of times.

[image]

Nelmajuva, 2021-11-28:
Additional: The connection once made, try again.

[image]

Taym95, 2021-11-28:
Check [the example app ](https://github.com/PhilipsHue/flutter_reactive_ble/blob/5edf819f1f304ca7c40cc4a73ff54a733a06a9af/example/lib/src/ble/ble_scanner.dart#L50), scanForDevices returns `StreamSubscription` cancelling the  StreamSubscription. cancel() will stops the scan.

closed 2021-11-28 by Nelmajuva