PhilipsHue/flutter_reactive_ble #463: Support on app communication with ble device

kostas95, 2021-12-01:
Hello, I'm trying to make my app communicate with a Beurer PO60 oximeter and i also have a documentation for the oximeter's transmission protocol, where many information is given but I'm having a hard time on implementing their instructions. 

I'm using a custom service for the PO60 that contains several characteristics. Inside the documentation, there's an instruction as below:

> To communicate with the custom service PO60 write commands are used to give the device information or ask for
> specific data. The write commands have to be sent to the custom characteristic with the UUID 0xFF01. The device is
> answering via notifications from the custom characteristic 0xFF02. The other characteristics in the service are not
> needed for the communication with the device. 

Those two services have the following properties:

[image]

The case here is that I don't know which `flutter_reactive_ble` functions I have to use in order to send a write command to the custom characteristic (with uuid 0xFF01) and then get the answering notifications from the custom characteristic (with uuid 0xFF02).

In previous implementations with other oximeter devices I was just using the `subscribeToCharacteristic(characteristic).listen()` function and it worked. In this case it doesn't. I've tried using the `writeCharacteristicWithResponse()` function but no data were transferred from the ble device to the app. Can anyone give me some hints on how to handle correctly these characteristics so I finally get some data to my app? Thank you for your time.

Taym95, 2021-12-01:
HI, even though this is not a library issue and could be a StackOverflow question because, but I will respond and please close the issues after that:

>>> The write commands have to be sent to the custom characteristic with the UUID 0xFF01.

Use this to send data: 
```
final characteristic = QualifiedCharacteristic(serviceId: serviceUuid, characteristicId: "0xFF01", deviceId: foundDeviceId);
flutterReactiveBle.writeCharacteristicWithoutResponse(characteristic, value: []);
``` 

>>> The device is answering via notifications from the custom characteristic 0xFF02.

Use this to subscribe to notification and read response:
```
final characteristic = QualifiedCharacteristic(serviceId: serviceUuid, characteristicId: "0xFF02", deviceId: foundDeviceId);
flutterReactiveBle.subscribeToCharacteristic(characteristic).listen((data) {
      // code to handle incoming data
    }, onError: (dynamic error) {
      // code to handle errors
    });
 ```

kostas95, 2021-12-02:
Thanks that's the correct way! For everyone that might have a same issue, I have to mention that in case of writing a characteristic the `value` parameters must be a list of `int` which every `int` represents a byte that tells the device what types of information must send.

closed 2021-12-02 by kostas95