PhilipsHue/flutter_reactive_ble #464: Mac OS M1 cpu build protobuf issue , I had fix it ,need update

cherho0, 2021-12-02:
protobuf platform error 

how to fix:
modify reactive_ble_mobile  build.gradle
protoc {

        if (project.hasProperty('protoc_platform')) {
            artifact = "com.google.protobuf:protoc:3.13.0:${protoc_platform}"
        } else {
            artifact = "com.google.protobuf:protoc:3.13.0"
        }
    }

add protoc_platform in gradle  :
open ~/.gradle
open gradle.properties
add protoc_platform=osx-x86_64   

hope update

remonh87, 2021-12-04:
@cherho0  I think you use an older version because we solved this in #396 . This fix is released in version `5.0.1`

closed 2021-12-04 by remonh87

cherho0, 2021-12-06:
[image]
I am sure it not work on may  m1 macbook , 5.0.1 and 5.0.2

remonh87, 2021-12-06:
It looks like it is fetching the correct binary but it is not there in the repo. Indeed I checked maven.alivun and the binary is not there. I do not know if this is a specific mirror for China but I am able to build the app here on my M1

cherho0, 2021-12-07:
It shoule be  "com.google.protobuf:protoc:3.13.0:osx-x86_64"   but not  osx-aarch_64.exe , I got your source code ,there is not the code  "osdetector.os == "osx"  here