PhilipsHue/flutter_reactive_ble #466: not an issue but just a demo

sonnny, 2021-12-03:
Hi,
     I have been trying other flutter ble plugin, but this particular flutter_reactive_ble I have found to be most reliable. Below is the youtube demo of android controlling multi train using dcc, flutter android app is using flutter_reactive_ble, source code is in the youtube description. 

[https://www.youtube.com/watch?v=wNRwCgRMJ1g](url)

Thanks,
Sonny

remonh87, 2021-12-04:
this looks really awesome!!  Thanks for sharing.

remonh87, 2021-12-04:
btw do you have a twitter handle? I would like to share this and mention you.

closed 2021-12-04 by remonh87

Taym95, 2021-12-04:
Great work, thanks for sharing 👏

sonnny, 2021-12-05:
> btw do you have a twitter handle? I would like to share this and mention you.

      My twitter handle is @sonnybalut

Thanks